ArduPilot/ardupilot #6639: Hardware recognition problem with kernel 4.9

256shadesofgrey, 2017-07-21:
#### Issue details
Ardupilot doesn't recognize the hardware correctly on kernel 4.9.35 and 4.9.38 RT (and probably others, but these are the 2 I tested with). It worked with kernel 4.4. 

Judging by contents of ardupilot/libraries/AP_HAL_Linux/Util.cpp , this has to do with the definitions starting at line 179 (_hw_names), since Util::get_hw_arm32() reads the processor info from /proc/cpuinfo, and with kernel 4.9 that file indicates the specific implementation of the chip (BCM2835) rather than the family (BCM2708). However just replacing the string seems to cause other issues (can't maintain connection with QGroundControl or APM Planner 2).

#### Version
Commit 6a01c6519708f3072c9192f6c438f25e3da459ce

#### Platform
Copter (tested), but probably all are affected.

#### Hardware type
Custom RPi 1 configuration, mostly based on pxfmini and a bit of navio.

#### Logs
$ sudo ./arducopter -A udp:192.168.0.77:14550
Could not detect RPi version, defaulting to 2/3

lucasdemarchi, 2017-07-21:
Please post the content of /proc/cpuinfo. Most of people moved to RPI 2/3 so it's possible we broke something for RPI 1.

karu2003, 2017-07-21:
I have the same error. With the new kernel (4.9.35) of RPi1 and RPi3, the same processor. BCM2835. :( Probably this is the kernel error.

karu2003, 2017-07-21:
lucasdemarchi --- "possible we broke something for RPI 1"  RPi Zero to?

lucasdemarchi, 2017-07-21:
> lucasdemarchi --- "possible we broke something for RPI 1" RPi Zero to?

Humn... Same processor is used on RPI zero. Please post the content of /proc/cpuinfo.

karu2003, 2017-07-21:
On Monday!!!

256shadesofgrey, 2017-07-22:
RPi 3
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>><
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 000000009b543d07
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>><

RPI 1 B+
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

pi@raspberrypilot ~ > cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 0010
Serial		: 00000000cb878783
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
RPI2 
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
pi@raspberrypilot ~ > cat /proc/cpuinfo 
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: BCM2835
Revision	: a21041
Serial		: 0000000001e7edeb
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>><
PRi ZERO
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 997.08
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 900093
Serial          : 00000000d015a49d
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

karu2003, 2017-07-22:
In the new kernel will be other.
/proc/device-tree/model   (human readable)
or 
/proc/cpuinfo  "Revision"   
Example of processing.
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-obtain_hw_model#L125-L364

peterbarker, 2021-07-30:
Looks like this was resolved in 2017.

closed 2021-07-30 by peterbarker