ArduPilot/ardupilot #6641: DSM input not responsive on BeagleBone Blue

jadonk, 2017-07-21:
#### Issue details
Could be user error, but it looks like the DSM code might not be enabled for the DSM input on BeagleBone Blue:

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_Linux/HAL_Linux_Class.cpp#L162

BeagleBone Blue has a DSM satellite receiver input connector on /dev/ttyS2. Using "-C /dev/ttyS2" doesn't seem to trigger listening to this input.

#### Version
3.5.0

Built and packaged by Robert Nelson (@RobertCNelson): [git-ardupilot-copter-blue](https://github.com/rcn-ee/repos/tree/master/git-ardupilot-copter-blue
)
#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
BeagleBone Blue

#### Logs
Not sure where to fetch any. Invoked with:

`/usr/bin/ardupilot/blue-arducopter -A udpin:0.0.0.0:14550 -C /dev/ttyS2 -l /var/APM/logs 2>&1 > /var/APM/logs/blue-arducopter.log &
`
My setup script:

[ardupilot_setup.sh](https://gist.github.com/jadonk/6080ca92d6e225eb89d33ad7744e1775)

RobertCNelson, 2017-07-21:
Here's the debian build settings from rules/control

rules:
```
###boards:
#board=bbbmini
board=blue

###targets:
target=copter
#target=plane
#target=rover

%:
	dh $@

override_dh_auto_configure:
	./waf configure --board=$(board) --no-submodule-update

override_dh_auto_build:
	./waf $(target)
	mv ./build/$(board)/bin/arducopter ./build/$(board)/bin/$(board)-arducopter
	mv ./build/$(board)/bin/arducopter-heli ./build/$(board)/bin/$(board)-arducopter-heli
```
Debian package depends:
```
Build-Depends:
	debhelper (>= 9),
	git,
	pkg-config,
	python-future,
	python-lxml,
```

Regards,

OXINARF, 2017-07-24:
@mirkix Can you answer this?

@jadonk @RobertCNelson This issues list is for feature requests and confirmed bugs, for user support please post in the forum (http://discuss.ardupilot.org). Since it is already here, I'll leave it open for a while to see if @mirkix answers. By the way, from your posts it looks like you are expecting to use one of the configurable serial ports to connect RC input, that's not how ArduPilot works. From my look at code, BB Blue is using the same AioPRU that BBBMini does. I assume you have read https://github.com/mirkix/ardupilotblue#receiver?

mirkix, 2017-07-24:
@jadonk @OXINARF The RC receiver signal has to be connected to connector E4 pin 4 of the BeagleBone Blue. This pin is a multi protocol pin (S.BUS, PPM-Sum, Spektrum Satellit DSM), the protocol will be automatically detected.

jadonk, 2017-07-25:
Is there a "I don't like the answer" option? We have a serial port labelled "DSM" on the board intended to provide this input with a connector that works off-the-shelf. I will make a new cable and recommend others do the same, but I don't like it.

closed 2017-07-25 by jadonk

mirkix, 2017-07-25:
When using the DSM labeled port we are limited to DSM protocol. Most users are using S.BUS or PPM protocols, that is the reason for using the multi protocol pin.

sherwial, 2017-11-21:
If we still want to use the DSM protocol, where would we enable this in the source? We have attempted to use E4 pin 4, but the RCInput example outputs "No channels". We have tried both E4 pin 4 and DSM with no results. I am having trouble finding where the RC port (Example: /dev/ttyS1) is specified, or where hal.rcin is located.