ArduPilot/ardupilot #6642: SCALED_PRESSURE.press_diff is incorrect

WickedShell, 2017-07-21:
#### Issue details
Scaled pressure is meant to be the compensated and scaled pressures for both the absolute (baro) and differential sensors (airspeed). We are currently sending the difference between the barometers current pressure and the ground pressure that the baro library is using for the differential value. https://github.com/ArduPilot/ardupilot/blob/979f6a1980f083fa1114ec9c44f06d8f6e0b39fb/libraries/GCS_MAVLink/GCS_Common.cpp#L1079

This is absolutely incorrect to the MAVLink spec http://mavlink.org/messages/common#SCALED_PRESSURE and means that there is no useful log on plane's of what the airspeed pressure actually is.

I realize we have been sending this message incorrectly for a long time, but this appears to be the only place to report the actual differential pressure, and frankly given that the ground pressure is available via a parameter I don't think that this would be to large of a breaking change to fix to the MAVLink spec. (At least there is no major normal part of the UI that should be relying upon this at the moment).

#### Version
Master

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

WickedShell, 2017-07-21:
@DonLakeFlyer @meee1 @billbonney would this cause any issue for you in the GCS anywhere if we corrected the press_diff field to comply with the spec?

DonLakeFlyer, 2017-07-22:
Current QGC only pulls the temperature field from that message to show to the user. Which is kind of odd, but someone else added that code and I guess they just need temp. So it has no affect on QGC.

meee1, 2017-07-22:
I vote correct it.

commit 6652dd73 (2018-03-21): GCS_MAVLink: Correct sending SCALED_PRESSURE.press_diff

Closes #6642
closed 2018-03-21 by OXINARF