ArduPilot/ardupilot #6644: Maxbotix i2c   fails in versioin apm 3.5 -- worked ok in previous version

jwgera, 2017-07-22:
#### Issue details
I upgraded to version 3.5, recalibrated, etc.  now my Maxbotix i2c sonar does not work.  
In mission planner, status -sonarrange I get values, but when I fly in alt hold mode the
sonar is ignored. 

attached is a screenprint of a flight where I ascended to about 1 meter and engaged alt hold
then flew over a hill that dropped about 10 feet. The copter stayed at same altitude but did
not follow the contour of the hill.

This was working fine in the previous version.

#### Version
3.5

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
quad_

#### Hardware type
pixhawk

#### Logs

[image]

OXINARF, 2017-07-24:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

closed 2017-07-24 by OXINARF

jwgera, 2017-07-24:
This looks like a bug to me.

OXINARF, 2017-07-24:
@jwgera Who confirmed it was a bug? It needs to be analyzed on the forum first (where you should post a log), then if it is indeed a bug an issue is created (or this one re-opened). I'm sorry to be strict on this, but our issues list is already enormous and many times what someone thinks is a bug is actually a bad configuration - not saying it is your case but can be.