ArduPilot/ardupilot #6647: SERVOx_FUNCTION	 changes don't work

ThomasH3, 2017-07-22:
#### Issue details
I've changed the servo function for Servo6 - Gimbal tilt and Servo5 - Yaw
Both don't work. Servo's are not receiving signals.

#### Version
3,5 Tricopter
#### Platform
[ x ] Copter
#### Airframe type
Y3
#### Hardware type
 pixracer

Pedals2Paddles, 2017-07-22:
I noticed the same last night. RC6_Function set for mount tilt input and that always works fine. You can watch CH6 in on the status and it is receiving the controller input.

SERVO6_FUNCTION set for mount tilt and servo 6 output is offline. SERVO6_FUNCTION set for RC Passthrough and it works properly.

OXINARF, 2017-07-24:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

I did a quick test here with a Cube and I can see the output value from ArduPilot changing. I don't have anything attached to check it is actually outputting the correct values though. Please post in the forum together with a log and it can be further analyzed. If a bug is found then we can re-open this.

closed 2017-07-24 by OXINARF