ArduPilot/ardupilot #6650: Copter: improve GPS Glitch handling and reporting separate from EKF failsafe

rmackay9, 2017-07-24:
With Copter-3.4.6 (or was it 3.3?) we moved the GPS glitch detection into the EKF.  The EKF certainly handles glitches better than the older inertial nav ever could but we've lost some reporting to the pilot.

We should add:

- pre-arm check that the GPS isn't glitching
- dataflash logging whenever a glitch occurs
- failsafe to allow momentary switching to AltHold if in Loiter, PosHold, etc.

OXINARF, 2017-07-24:
Related to #6063. I think @priseborough has a good idea there with AHRS providing an error estimate of position and velocity, then vehicles can decide what to do based on that.

I think that going back to a specific GPS glitch failsafe isn't the best idea, but a position failsafe would certainly be worthy - the reason being that we can lose GPS but still have good position estimate if, for example, we have optical flow installed.

Also, I'm not sure about backporting this to 3.5, I think it needs quite a bit of testing before putting it in a stable release.

peterbarker, 2021-07-30:
I've updated the initial post for this one as we have 2 of the three items done.

Haven't got custom GPS-glitch failsafe actions yet.

rmackay9, 2022-04-13:
AltHold is an option in FS_EKF_FAILSAFE so I think we can close this.

closed 2022-04-13 by rmackay9