ArduPilot/ardupilot #6652: Plane, vectored Tailsitter, Q_M_THST_HOVER has no effect in Q_STABILIZE

lorbass, 2017-07-24:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

tridge, 2017-07-26:
indeed, Q_M_THST_HOVR is not expected to do anything in QSTABILIZE. It is only used when we are in an altitude hold mode.
We could add input curves in quadplanes, but I'm not totally sure we want to, especially with hover learning. If it learnt a bad hover value then it could become hard to control

tridge, 2017-07-26:
I'll close as this is expected behaviour

closed 2017-07-26 by tridge

lorbass, 2017-07-26:
Thank you for this expanation. Otto