ArduPilot/ardupilot #6661: Copter: in angle control GUIDED mode the drone keeps rotating on its yaw axis when told not to do so

deb0ch, 2017-07-24:
#### Issue details
In GUIDED mode, using angle control with the `SET_ATTITUDE_TARGET` messages, the drone keeps rotating on its yaw axis.

It will first quickly go to a yaw angle (presumably the one set in the command's quaternion), and then will keep turning at a more or less constant rotation speed.

During another flight we even had a crash right when we passed the vehicle from LOITER to GUIDED, using the included code example with the `yaw_rate` at `2` in the parameters of `send_attitude_target`: the drone leaned very hard (maybe 45°) on both its pitch and roll axes, quickly losing altitude.

The `WP_YAW_BEHAVIOR` was set to `0` (`WP_YAW_BEHAVIOR_NONE`) for all those tests.

I took a couple videos using the example code below to illustrate the problem, I will include them if requested. Same with the logs, will try to find them if requested :smile: 

#### Version
3.4.6

#### Platform
- [ ] All
- [ ] AntennaTracker
- [x] Copter
- [ ] Plane
- [ ] Rover

#### Airframe type
quad

#### Hardware type
Erle Brain 3

### Code

``` python

import math
import time

import dronekit

copter = dronekit.connect('udp:10.0.0.42:5762', wait_ready=True)

def send_attitude_target(self, roll_angle=0.0, pitch_angle=0.0,
                         yaw_angle=0.0, yaw_rate=10, thrust=0.5):
    msg = copter.message_factory.set_attitude_target_encode(
        0,
        0,                                         # target system
        0,                                         # target component
        0b00000000,                                # type mask: bit 1 is LSB
        to_quaternion(roll_angle, pitch_angle, yaw_angle),    # q
        0,                                         # body roll rate in radian
        0,                                         # body pitch rate in radian
        yaw_rate,                                  # body yaw rate in radian
        thrust)                                    # thrust
    copter.send_mavlink(msg)

def to_quaternion(self, roll=0.0, pitch=0.0, yaw=0.0):
    """Convert degrees to quaternions."""
    t0 = math.cos(yaw * 0.5)
    t1 = math.sin(yaw * 0.5)
    t2 = math.cos(roll * 0.5)
    t3 = math.sin(roll * 0.5)
    t4 = math.cos(pitch * 0.5)
    t5 = math.sin(pitch * 0.5)
    w = t0 * t2 * t4 + t1 * t3 * t5
    x = t0 * t3 * t4 - t1 * t2 * t5
    y = t0 * t2 * t5 + t1 * t3 * t4
    z = t1 * t2 * t4 - t0 * t3 * t5
    return [w, x, y, z]

while True:
    if(copter and copter.mode == dronekit.VehicleMode("GUIDED")):
        send_attitude_target(0,  # roll
                             0,  # pitch
                             0,  # yaw_angle
                             0,  # yaw_rate
                             0.5)  # thrust
        print("attitude command sent")
    time.sleep(0.1)
```

OXINARF, 2017-07-25:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

I hope that the to_quaternion comment is just a mistake and you are passing it radians and not degrees.

Anyway, your script is wrong with those self in the function definitions, they don't belong to any class so there is no self to be passed - which ends up making yaw_rate be 0.5 in your example. Fixing that, I couldn't reproduce your issue with that piece of code.
If you have code that actually reproduces your issue, post here and I'll try it. For now, I'm closing this.

closed 2017-07-25 by OXINARF

deb0ch, 2017-08-09:
Sorry for this messy and misplaced issue report, it was the end of the day and we were pretty tired and desperate (we've been struggling on issues with the entire apm ecosystem for months now).

We finally got it working, we were confused reading the docs about how the SET_ATTITUDE_TARGET message worked.

We were thinking that the 3rd bit of the bitmask was to ignore the yaw altogether OR set the absolute yaw using a particular rate, but after reading Ardupilot code we figured out that the bitmask was actually to EITHER use the absolute setpoint OR the specified rate.

Maybe it would be worthy to clarify it even more in the MAVLINK specification (I haven't seen it documented anywhere else).

Yes there was indeed some errors in the example I gave, we tried to extract this minimal example from our codebase for reporting but forgot about the `self`s. It was reproducing exactly our issue when we tested it, which is why we didn't see it. The quaternion takes radians indeed, the comment is wrong.

So yes, in the end this was not really an issue with Ardupilot but rather with our understanding of the docs.