ArduPilot/ardupilot #6665: no RTK in ardusub 3.5

itayguy, 2017-07-25:
#### Issue details
I cannot get rtk float or fix in ardusub 3.5
#### Version
ardusub 3.5 / 3.6-dev

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[x] Sub

#### Airframe type
6 vector
#### Hardware type
pixhawk v2

#### Logs
_Please provide a link to any relevant logs that show the issue_

WickedShell, 2017-07-25:
@jaxxzer to resolve this all you need to do is copy this block https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1627-L1634 into ArduSub's mavlink and then replace the copter reference with sub :) (Unless @peterbarker beats you to it and makes it a common helper before you get to it)

peterbarker, 2017-07-25:
I hope you just didn't use the word "copy".  We do try to keep the use of foul language here to a minimum, right?

WickedShell, 2017-07-25:
Sorry, sorry. You should yank it into your buffer and then paste it where needed!

peterbarker, 2017-07-25:
https://github.com/ArduPilot/ardupilot/pull/6667

peterbarker, 2017-07-25:
I can't really test that one - @itayguy any chance you could test this PR?

OXINARF, 2017-07-25:
#6667 was merged so Sub should now be able to get into RTK status.

closed 2017-07-25 by OXINARF

jmachuca77, 2017-07-25:
Obvious question... but you are not trying to get an RTK lock underwater right? Or can you?

jaxxzer, 2017-07-25:
You can't get a lock underwater with any GPS antenna. I don't know what @itayguy plans are, but some people operate ArduSub at the surface in favor of ArduBoat.

Thanks for taking care of this @WickedShell and others.

WickedShell, 2017-07-25:
The other advantage of the patch that was merged is GPS MAVLink injection is now available so you could send GPS_INPUT to the vehicle http://mavlink.org/messages/common#GPS_INPUT

jaxxzer, 2017-07-25:
Weird, I've been using GPS_INPUT for months on Sub, usually Sub 3.5.

WickedShell, 2017-07-25:
@jaxxzer oops @peterbarker didn't nuke the explict handling of that message in sub.

itayguy, 2017-07-25:
I was offline a bit, thanks all for your quick response  - about RTK underwater - should work fine , if u put your base antenna and ground station under water :))

itayguy, 2017-07-26:
can't test ur patch since it's too head from sub3.5 , and ur master code cannot compile ...