ArduPilot/ardupilot #6670: support I2C and analog airspeed sensors on pixhawk

geofrancis, 2017-07-25:
pixhawk currently can use either the analog sensor or i2c but not both, i have had a few planes go down over the years now due to airspeed malfunctions,

loose wire making it think it was doing 200mph+
blocked and partially blocked pitot tube
cracked tube between pitot and sensor 
water drops. 

for probably one of the most critical sensors on a plane i find it unbelievable that there is no redundancy or failsafe for bad data from the airspeed sensor only switching if there is no data.

both sensors are readily available, they just need to be active at the same time and a way of disabling the airspeed data if there is too high a variance in the reported speeds.

sashan13, 2017-09-01:
Agree very much. At least two are required. We have 2 3.3v adc inputs, why not use them? Working analog sensor is better than failed digital one. 
At least something like majority vote with  sensors should be possible for larger planes.

EShamaev, 2017-09-17:
Can take on that. I will have a plane with 2 UAVCAN pitot sensors very soon.

geofrancis, 2017-09-18:
Very cool I never thought about comparing to estimated airspeed.

magicrub, 2018-01-03:
submitted PR: https://github.com/ArduPilot/ardupilot/pull/7174

WickedShell, 2018-04-26:
Dual airspeed support is in, #7174. Airspeed failure detection/mitigation is an ongoing issue that is being tracked elsewhere.

closed 2018-04-26 by WickedShell