ArduPilot/ardupilot #6677: Creating Vagrant 404 file not founds

pawelsoon, 2017-07-26:
When trying to install SITL on OSX using Vagrant (command $vagrant up) I get error message:

peterbarker, 2017-07-27:
Try `vagrant box update`

peterbarker, 2017-12-19:
I moved our Vagrant boxes forward because of this.

closed 2017-12-19 by peterbarker

KenanBektas, 2018-10-16:
Dear @peterbarker I receive the same 404 error while trying to work with the branch 3.4. Actually the master (3.4.6) seems to have the same problem because the Vagrantfile includes config.vm.box = "ubuntu/yakkety32". Can you help/comment on that?

KenanBektas, 2018-10-19:
Dear @peterbarker your suggestion worked. Thank you for that. Unfortunately, I have now a python related problem. The following script:
vagrant ssh -c "sim_vehicle.py -j 2"
creates the following message:
bash: sim_vehicle.py: command not found. Do you have any ideas to solve this issue?

peterbarker, 2018-10-19:
On Fri, 19 Oct 2018, KenanBektas wrote:

> Dear @peterbarker your suggestion worked. Thank you for that. Unfortunately, I have now a python related problem. The following script:
> vagrant ssh -c "sim_vehicle.py -j 2"
> creates the following message:
> bash: sim_vehicle.py: command not found. Do you have any ideas to solve this issue?

Can you paste the output of the machine bring-up?

It looks like your path is incorrect - it's possible the machine didn't 
come up correctly.

If you get a shell on the machine you can possibly:
cd /vagrant
./Tools/autotest/sim_vehicle.py -v ArduCopter

KenanBektas, 2018-10-19:
g++ was also missing so I installed that as well. Then, as you suggested, I run  
sim_vehicle.py -v ArduCopter
That created a log including
'configure' finished successfully (1.680s)
but also the following

Traceback (most recent call last):
  File "/vagrant/modules/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/vagrant/modules/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
    from future import standard_library
ImportError: No module named future

mavgen returned 1 error code
Waf: Leaving directory `/vagrant/build/sitl'
Build failed
 -> task in 'mavlink' failed (exit status 1): 
	{task 3073901748L: mavgen ardupilotmega.xml -> }
''
SIM_VEHICLE: Build failed
SIM_VEHICLE: Killing tasks
Connection to 127.0.0.1 closed.

KenanBektas, 2018-10-19:
Thank you very much. This is what I get:

+ echo 'Initial setup of SITL-vagrant instance.'
Initial setup of SITL-vagrant instance.
+ BASE_PKGS='gawk make git arduino-core curl'
+ SITL_PKGS='g++ python-pip python-matplotlib python-serial python-wxgtk3.0 python-scipy python-opencv python-numpy python-empy python-pyparsing ccache'
+ PYTHON_PKGS='pymavlink MAVProxy droneapi future'
+ PX4_PKGS='python-serial python-argparse openocd flex bison libncurses5-dev           autoconf texinfo build-essential libftdi-dev libtool zlib1g-dev           zip genromfs cmake'
+ UBUNTU64_PKGS='libc6:i386 libgcc1:i386 gcc-4.9-base:i386 libstdc++5:i386 libstdc++6:i386'
+ ARM_ROOT=gcc-arm-none-eabi-4_9-2015q3
+ ARM_TARBALL=gcc-arm-none-eabi-4_9-2015q3-20150921-linux.tar.bz2
+ ARM_TARBALL_URL=http://firmware.ardupilot.org/Tools/PX4-tools/gcc-arm-none-eabi-4_9-2015q3-20150921-linux.tar.bz2
+ ARDUPILOT_TOOLS=ardupilot/Tools/autotest
+ VAGRANT_USER=ubuntu
+ usermod -a -G dialout ubuntu
usermod: Permission denied.
usermod: cannot lock /etc/passwd; try again later.

KenanBektas, 2018-10-19:
@peterbarker I went one more step and was able to run 
sudo ./initvagrant.sh 
The problem that I have reads as follows: 
....
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package g++-4.7
E: Couldn't find any package by glob 'g++-4.7'
E: Couldn't find any package by regex 'g++-4.7'

peterbarker, 2018-10-19:
On Fri, 19 Oct 2018, KenanBektas wrote:

> @peterbarker I went one more step and was able to run
> sudo ./initvagrant.sh
> The problem that I have reads as follows:
> ....
> Reading package lists... Done
> Building dependency tree
> Reading state [...]

Great.  So now we're at the core of the problem - the one I mentioned 
initially - that the Vagrantfile and the Vagrant init file aren't a good 
match.

The machine you've brought up *probably* only has a later version of g++ 
available - you can work that out with something like "apt-cache search 
g++".  It might be you could use a newer g++-4.7 in place of the old one.

Another option is to try progressively older versions of Ubuntu present in 
the Vagrantfile - the set should be relatively complete, including ones 
you can no longer download (there so that people with existing machines 
that old can still use them!).

Another option might be to replace with initvagrant.sh and 
install-prereqs-ubuntu.sh with their equivalents from the master branch. 
If that works then it might be a good way forward for the project to keep 
Vagrant working on the stable/released branch; replace all three files in 
all active branches when any are updated in master.