ArduPilot/ardupilot #6679: ArduSub compile error: 'class GCS' has no member named 'send_message'

itayguy, 2017-07-26:
#### Issue details
 error: 'class GCS' has no member named 'send_message'
ArduSub/GCS_Mavlink.cpp:11:11: 

#### Version
.dev
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[X] Sub

#### Airframe type

#### Hardware type

#### Logs

jaxxzer, 2017-07-26:
Builds for me, and https://travis-ci.org/ArduPilot/ardupilot/builds/257295368. Try `make clean` and `make px4-clean` from the ArduSub directory.

closed 2017-07-26 by jaxxzer

itayguy, 2017-07-28:
px4-v2

peterbarker, 2017-07-28:
On Fri, 28 Jul 2017, itayguy wrote:

> px4-v2

Unable to replicate.

Please try doing a "make px4-clean" and retry.

If using waf, please execute a "waf clean" and retry.

If it still fails - what git hash are you trying to compile?

itayguy, 2017-07-28:
it works thanks!