ArduPilot/ardupilot #6680: Support for quick tweaking (profiling) of some settings

SergeyBokhantsev, 2017-07-26:
#### Issue details
As many of another pilots I do have several batteries of different capacity, weight, voltage etc.
So for the one battery I want to use one settings (capacity, failsafe voltage, failsafe behavior, even PIDs)
For the another one I must use different settings.

Or as another example is when someone are using different props and alters them. So he must use different PIDs as well.

It would be great to have a possibility to tweak some settings basing on the pre-configured profiles. 
Important point here is it must be doable without connecting to notebook/phone. It must be simple and quick to select the proper configuration right on a flight site.

Technically one of the option could be:
- have stored profiles on flash card. This could be a text file like: 
`{
Profile: 1,  { ... settings to tweak ... }
Profile: 2,  { ... settings to tweak ... }
}`
- when vehicle is disarmed and some RC stick combination is pushed then profile 1 applied. That must be indicate by led, buzzer and mavlink message. Next time pushing that RC stick combination must apply next profile and so on.

#### Platform
[ x ] Copter

amilcarlucas, 2017-10-10:
Just use a tablet. :P

CodingGhost, 2019-12-23:
I think the Idea of just having a parameter list as file on the SD and loading it via a stick command (only when disarmed), then seeing it as an OSD/Mavlink message is a really good Idea. It shouldnt be too hard to implement, as you wouldnt have to alter all the code. This would also solve my #13093

Pedals2Paddles, 2019-12-23:
I like stick gestures. But the majority of the dev team does not for a variety of valid reasons. Stick gestures that load parameter changes is highly unlikely to be supported.  What you're describing can be easily accomplished by loading a parameter file with the GCS. Is there a reason you do not wish to do that? A companion computer could automate an out of range scenario.

CodingGhost, 2019-12-23:
I dont like this, because this means I would always have to carry a laptop around with me + it takes time.
I dont have enough space for a companion computer on my craft. Also I think its overkill to have a full raspberry just to change parameters.

IamPete1, 2022-08-05:
This can be done with scripting.

IamPete1, 2022-08-05:
We also have param edit via the OSD.