ArduPilot/ardupilot #6690: Plane: expand PID min/max ranges

magicrub, 2017-07-28:
Plane AUTOTUNE is capable of creating values that are beyond the suggested min/max values. The min/max values probably need to be pushed out a little more.

Reported by Jacques Eloff on Facebook:
https://www.facebook.com/groups/pixhawk2/permalink/1370993012983317/?comment_id=1372645929484692&comment_tracking=%7B%22tn%22%3A%22R%22%7D

Naterater, 2019-01-04:
I changed the D values here: https://github.com/ArduPilot/ardupilot/pull/10082. Once merged, this should be solved.

WickedShell, 2019-04-09:
Some changes have happened here, and this looks to be the kind of issue that has no clear action to take, so I'm going to close this at this point, and if we have specific parameters to improve we should be opening specific issues, rather then this one which is hard to execute on because it's so general.

closed 2019-04-09 by WickedShell