ArduPilot/ardupilot #6691: Plane: Takeoff course hold doesn't account for the effect of wind

WickedShell, 2017-07-28:
#### Issue details
Plane takeoffs currently lock a desired course and trusts the navigation controller to manage that. However because the aircraft only cares about flying the desired navigation course any error that occurs (such as being pushed off the ground track due to winds) is not compensated for. This is not an issue when taking off into the wind, but any crosswind component won't be compensated, so any takeoff with a crosswind, or that encounters shifting winds (IE as the vehicle clears an obstruction) will see a sudden and unexpected course shift.

This screenshot is from SITL with a 8 m/s crosswind, the hard left at the end is the vehicle shifting to the next waypoint, however there is a 50 meter cross track error from the orginal takeoff before this happens.

[image]

My proposed solution is to instead project a fake waypoint forward from the current location and fly that path out. (Re extending the fake waypoint as required if we haven't hit the target altitude by then).

#### Version
Master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

magicrub, 2017-07-30:
This was discussed in the past and the argument for the current design was for the takeoff to prioritize gaining altitude and airspeed over anything else. Heading hold best effort was just a stability requirement. 

I believe that is still a correct goal...

However, we hold that requirement until the altitude is met and that is the problem. Way too long. 

Once cruise airspeed has been met and a reasonable altitude gain of 10m or so then yes, it's a good idea to cross track to a virtual wp a few km be away

Naterater, 2019-01-04:
I think that the WP should be active at all times, but just honor LEVEL_ROLL_LIMIT until TRIM_ARSPD_CM is met.

magicrub, 2019-01-04:
possibly...