ArduPilot/ardupilot #6692: Fr-Sky Telemetry "SPORT" (protocol 10) doesn't work on serial console (CONS on Pixhawk 2)

robustini, 2017-07-29:
#### Issue details
It would be interesting to use the serial 5 "CONS" of "Pixhawk 2.1 with protocol 10 (Fr-Sky SPORT serial) for the telemetry on Taranis.
Having two gps and Edison i can't connect it anywhere else, and I would like to avoid using a parallel connection on TELEM1.

#### Version
Copter V3.5

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
X6

#### Hardware type
Pixhawk 2.1, Taranis Plus, rx X8R with the "FlightDeck - Craft and Theory" telemetry on SPORT

#### Logs

lvale, 2017-07-29:
@robustini Better yet (and perhaps easier) would be having ALL the physical serial ports available to any protocol available on serial ports.

I might be doing a huge error, but having also NSH available on any serial port as an option would be the best solution

robustini, 2017-07-30:
@lvale I agree!

lvale, 2017-07-30:
Added as topic for next dev call but I would aim for adding any serial to any protocol

proficnc, 2017-08-01:
Added this as a proposal as well to see if we can get any takers on doing this.
https://discuss.ardupilot.org/t/enable-serial-5-for-mavlink-and-craft-and-theory-etc/19629

robustini, 2017-08-01:
Thanks @proficnc !
I don't think many people today need to fly using such a complex configuration, but one thing is certain: a test pilot doesn't need the "NSH console" for its flight tests, but of course useful for developers.
A parameter to enable/disable it would be very useful, and perhaps have the ability as rightly wrote @lvale  to make it active in other serials.

robustini, 2017-08-01:
For example with a bitmask:

NSH_Console, 0   (disabled)
NSH_Console, 1   (enabled on serial CONS)
NSH_Console, 0   (enabled on GPS2 port)
NSH_Console, 0   (enabled on TELEM2 port)

etc. etc. .....

CharlieRome, 2017-08-07:
Thank you all.
Yes, I agree, we are in dire need of an extra Serial Port. My setup is very similar to Marco's:
1) telemetry radio
2) Raspberry Pi as a companion computer (it will soon be a nVidia Jetson TX1 or 2).
3 and 4) two GPSs for redundancy and blending.
I absolutely need to add an ADS-B receiver receiver to my configuration.

Thank you for taking care of this.

Keep up your wonderful work!

tridge, 2017-08-13:
I've implemented this here:
https://github.com/ArduPilot/ardupilot/pull/6761

closed 2017-08-15 by robustini