ArduPilot/ardupilot #6711: Copter: WPNAV_SPEED not obeyed in Guided Mode

karthikdesai, 2017-08-01:
#### Issue details
This test was conducted on SITL on master
The flight was automated where the copter flies between 2 points at constant speeds. The param WPNAV_SPEED is set every time it reaches the destination point before it starts with the next flight path. The param is increased by 50 cm/sec after reaching every point, all in Guided mode. So for example, we have A, B, C, D points. Copter flies from A to B at 1 m/s. B to C at 1.5 m/s and so on.

Initially, before the whole 'mission' script takes over, the` WPNAV_SPEED was set to 200`. There are 12 points. Then in the script, before the first point, the WPNAV_SPEED is set to 100 (in guided mode). This maximum speed is obeyed. When the WPNAV_SPEED is increased to >200, the copter continues to fly at maximum 2 m/s. The log file shows the WPNAV_SPEED was increased upto 650. 
We experimented this by setting WPNAV_SPEED to 500 initially. And the copter flies correctly till it reaches the point where it supposed to fly at this speed. The copter continues to fly at this speed after the point, which seems to be its maximum speed, eventhough the param WPNAV_SPEED was set to >500 (550, 600 and 650).

So what the user sets in the beginning seems to be the maximum speed the copter will fly even if the WPNAV_SPEED param is increased than the initially set value

#### Version
APM:Copter V3.6-dev (948e9056)

#### Platform
[  ] All
[  ] AntennaTracker
[* ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
OCTA_QUAD

#### Hardware type
SITL

#### Logs
https://gist.github.com/karthikdesai/570d08d4d88f2740e633b16f34997f61#file-guided_mode_test-log

karthikdesai, 2017-08-01:
This issue can be reproduced on SITL by setting an initial WPNAV_SPEED to say 200. Then flying in guided mode and changing the value to some thing lesser than 200 is going to obeyed. But if the value is set to more than 200, the speed is still stuck at 2 m/s and the write confirmation of WPNAV_SPEED value can be checked!
@rmackay9 Is this normal functioning in guided mode?

rajat2004, 2019-04-17:
I just tested this in SITL, after setting `WPNAV_SPEED` for the first time, the value is obeyed but after this, changing the param doesn't have any effect on the speed.
I think that this is the normal functioning but it will be great if someone can confirm this.

Using `setspeed` command or `MAV_CMD_DO_CHANGE_SPEED` seems to work and changes the speed in between the mission

rajat2004, 2019-04-17:
One issue might be the write confirmation of `WPNAV_SPEED` param value

rmackay9, 2019-05-16:
This is really the same as this issue: https://github.com/ArduPilot/ardupilot/issues/6193

amilcarlucas, 2021-03-09:
Hey Karthik. I think this runs fine in 4.0 and got broken again in master. We should test it at some point.

rmackay9, 2021-03-09:
I think this will all work fine with SCurves but instead of changing the WPNAV_SPEED parameter users should use the do-change-speed command which can be sent from MP and other GCSs while the vehicle is flying.

IamPete1, 2021-08-15:
Sounds like this is fixed with SCurves.

closed 2021-08-15 by IamPete1