ArduPilot/ardupilot #6712: Uavionix ADSB Out issue (SITL)

flyingsilverfin, 2017-08-01:
#### Issue details
I'm attempting to get ArduCopter to use ADSB-Out in a SITL build. There's an existing SIM_ADSB module which can receive sent ADSB messages from ArduCopter. However, there's an issue somewhere in the abstractions layers. To make sense of it I've posted the relevant part of the stack trace below:

```
 #0 _mav_finalize_message_chan_send (chan=chan@entry=MAVLINK_COMM_1, msgid=msgid@entry=10002, packet=packet@entry=0x7ffffffddde0 "\275G\200Y\240\376\353\352\377\316\350X\220\351\b", min_length=min_length@entry=41 ')',  length=length@entry=41 ')', crc_extra=crc_extra@entry=186 '\272')
at /tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v2.0/mavlink_helpers.h:293
#1 mavlink_msg_uavionix_adsb_out_dynamic_send (squawk=1200, state=8, emergencyStatus=0 '\000', VelEW=0, velNS=0,velVert=0, accuracyVel=40, accuracyVert=20, accuracyHor=200, baroAltMSL=607047, numSats=10 '\n', gpsFix=6 '\006', gpsAlt=584080,
gpsLon=1491652351, gpsLat=-353632608, utcTime=1501579197, chan=MAVLINK_COMM_1)
at /tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v2.0/ardupilotmega/../uAvionix/./mavlink_msg_uavionix_adsb_out_dynamic.h:321
#2  AP_ADSB::send_dynamic_out (this=this@entry=0x77ce38 <copter+14296>, chan=chan@entry=MAVLINK_COMM_1)
at /home/js/ardupilot/libraries/AP_ADSB/AP_ADSB.cpp:539
#3  AP_ADSB::update (this=this@entry=0x77ce38 <copter+14296>)
at /home/js/ardupilot/libraries/AP_ADSB/AP_ADSB.cpp:249
```

As we can see the `AP:ADSB::update` calls a `send_dynamic_out` which uses `mavlink_msg_uavionix_adsb_out_dynamic_send`. Now this function attaches a Message ID of 10002 for a dynamic out. Unfortunately, the final `_mav_finalize_message_chan_send` has a condition:

```
if (msgid > 255) {
                // can't send 16 bit messages
                _mav_parse_error(status);
                return;
}
```

which results in the ADSB-Out message being discarded.

I have a couple of relevant questions. For one, am I safe to replace the message ID 10002 with 246, which is the standard ADSB_Vehicle message ID and seems to have the same data? Or is it better to add support for 16 bit messages? This seems like it might be involved/impossible as the header size is fixed? 

Lastly, I'm not sure why Uavionix used these IDs, but none of the code I'm looking at differs in SITL versus actual flight situations (I think), so I think the transceiver OUT capabilities would not work in practice either? Has this been tested (I would if the devices weren't so expensive)?

Other infos:
Version: Copter 3.5 (SITl quad). Probably same error in plane and other frames.

magicrub, 2017-08-01:
For some reason you're building MAVLink v1.0. the adsb out driver requires MAVLink v2 which has been the default for ArduPilot for a Year.

flyingsilverfin, 2017-08-01:
Ah that might be it, I'm finding the build options generally quite confusing - how would I specify the MAVLink version? Also, how did you identify that I've got 1.0 not 2.0? It's using the 2.0 header files in the stack trace.

flyingsilverfin, 2017-08-01:
Perhaps it has something to do with the fact that i was running 

> make sitl -j4

to create the ELF file? I found that this was the executable I could attach GDB to (the build/sitl/bin/arducopter binary doesn't have any debugging symbols, even when running waf with --debug).

magicrub, 2017-08-01:
MAVLink 2 uses a 24bit msg id. MAVLink 1 is 8bit.

All Linux builds, including SITL, require using waf. Make is depreciated. Make is only still available for stm32 builds.

Try running:
sim_vehiclel.py -j4

Closing this issue. Please post in discuss.ardupilot.org if You need additional help running SITL.

closed 2017-08-01 by magicrub

OXINARF, 2017-08-01:
This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

That piece of code from `_mav_finalize_message_chan_send` is only reached when the channel you are sending to is sending MAVLink1. So yes, the message will be discarded since you don't have MAVLink2 active in that link and that ADSB message is a MAVLink2-only message.

> For one, am I safe to replace the message ID 10002 with 246, which is the standard ADSB_Vehicle message ID and seems to have the same data?

I'm not sure where you are seeing they have the same data, but you can do whatever you want to code, it just won't make it to our repository as it would break the uAvionix driver.

> Or is it better to add support for 16 bit messages?

It is already there, you just need to use MAVLink2, MAVLink1 doesn't support 16 bit IDs.

> Lastly, I'm not sure why Uavionix used these IDs

Two main reasons:
- we have few 8 bit IDs left
- we decided it was an excellent opportunity to give a manufacturer a range of IDs for the first time. Any other manufacturer may want to have a message with less or more or different parameters and they'll be able to get their ID range and their messages

You might wonder why ADSB-in wasn't done the same way and the main reason is that MAVLink2 wasn't available yet at the time it was done.

> so I think the transceiver OUT capabilities would not work in practice either? Has this been tested (I would if the devices weren't so expensive)?

Yes, it was tested and it works.

OXINARF, 2017-08-01:
Wow, thank you GitHub for not updating the damn page! I was late to the party 🙂

flyingsilverfin, 2017-08-02:
Ah thanks, I've also finally got my head around what that's actually doing and all the waf options. If I may ask one more question while we're here - where is the default list of parameters stored? I found that SERIAL1_PROTOCOL is set to 1 when I start the instance - I can just put this in a startup command but would be easier if it's set correctly to start with.

OXINARF, 2017-08-02:
Well, it depends on what you are doing:

- defaults are set in code when parameters are defined
- when running SITL binary directly, you can use the option `--defaults` to give a path to a file with default parameter values
- when running SITL with sim_vehicle.py, you'll need to edit either the scripts to change the default loaded file or change the default files themselves