ArduPilot/ardupilot #6724: Copter/QuadPlane: parameter to define the motor ramp curve during takeoff

robustini, 2017-08-04:
#### Issue details
For automatic takeoff during AUTO/PosHold/Alt-Holde/Loiter flight mode, in some configurations it would be necessary tune better the ramp speed curve of the motors.
During this phase the Alt-Hold intertial controller comes into play, and with some hardware setup doesn't work very well, sometimes the drone takes off too fast and tilted.
And some ESCs don't like a pwm ramp too fast, some go to desync.
So what I think is missing is a parameter to define the speed curve of this ramp, or a better management of the Alt-Hold controller, even in the land phases.
"MOT_SPOOL_TIME" and "MOT_SPIN_ARM" don't solve the problem, they have to act on something else, not in what I mean.

#### Version
Copter V3.5.1
Plane V3.8.0 RC6

#### Platform
[ x ] Copter
[ x ] QuadPlane

#### Airframe type
Is irrelevant

#### Hardware type
Is irrelevant

#### Logs
Available on request, also video documentation

robustini, 2017-08-05:
Please @OXINARF can you add also the "Copter" label to this issue?
Thanks!

OXINARF, 2017-08-05:
@robustini Done 🙂

lvale, 2017-09-07:
discussed this with @robustini.

robustini, 2017-09-07:
Thanks @lvale!

tridge, 2017-09-07:
Why isn't MOT_SPOOL_TIME what you want?

robustini, 2017-09-08:
Because "MOT_SPOOL_TIME" is the time in seconds to spool up the motors from zero to min throttle, and not "min throttle to takeoff throttle".

tridge, 2017-09-11:
> Because "MOT_SPOOL_TIME" is the time in seconds to spool up the motors from zero to min throttle (MOT_SPIN_ARM), and not "min throttle to takeoff throttle".

its actually both. It controls both the rise time to the min throttle, plus also a rise time from there to full throttle control. 
Here is a graph of a LOITER takeoff on a quad with MOT_SPOOL_TIME=5
[image]
you can see the initial 5 second ramp to min throttle, then a subsequent ramp (also of 5 seconds) to full throttle.

tridge, 2017-09-12:
the documentation does need an update though

tridge, 2017-09-12:
on the call we discussed adding a MOT_SPOOL_TIME_H, default to 0. If it is zero it uses the MOT_SPOOL_TIME (so current behavior). Otherwise it controls the 2nd half of the ramp

robustini, 2017-09-12:
Thanks Andrew, i wait the new parameter!

robustini, 2017-10-25:
@tridge, no news about that?

maciek01, 2018-10-28:
Hi . Any update on this? I have the same issue. Motors spool up fine in stabilized mode but in any of the autonomous modes mentioned above 4 out of 6 ESCs are desyncing and motors shut off.
The only way I can use loiter is takeoff in stabilized then switch the mode. Things were fine with xrotor hobbywing 40a. Then I switched to 50a red bricks (https://www.banggood.com/Red-Brick-50A-ESC-Brushless-ESC-OPTO-NO-BEC-50AOPTO-p-77969.html?utm_design=41&utm_source=emarsys&utm_medium=Shipoutinform171129&utm_campaign=trigger-emarsys&utm_content=Winna&sc_src=email_2671705&sc_eh=19ca7d4d939002cd1&sc_llid=7184001&sc_lid=104858042&sc_uid=mWbUCsvi4i) and started facing this issue.

chobitsfan, 2022-07-08:
> its actually both. It controls both the rise time to the min throttle, plus also a rise time from there to full throttle control.
Here is a graph of a LOITER takeoff on a quad with MOT_SPOOL_TIME=5

I tested it with current master, CubeBlack, arm and takeoff in guided mode, MOT_SPOOL_TIME = 3

[image]

It takes 3 sec to min throttle, but only 1 sec to full throttle. I look at code and it seems MOT_SPOOL_TIME is only use in SPOOLING_UP stage ?

lthall, 2022-07-08:
I didn't know about his issue but this is my next development project is to specifically handle the transition between landed and flying. The plan is to provide a throttle ramp time and then detect the moment we start to climb and then switch to the Alt_Hold controller for vertical control.

This will give the users direct control over the take-off transition. It will also handle variable take off weight more elegantly. Finally and most importantly it will mean the take-off behaviour is independent of the vertical height controllers.

lthall, 2022-07-10:
This is the pull request that should resolve this issue if anybody wants to take a look. #21158

lthall, 2022-11-21:
I think this has been addressed and merged. We can open it again if I am mistaken.

closed 2022-11-21 by lthall