ArduPilot/ardupilot #6731: Suggest TECS control parameter K_STE2Thr multiply (THRmaxf-THRminf) term

nocternal, 2017-08-06:
#### Issue details
Learning and looking into TECS control law, find that 
K_STE2Thr = 1 / (timeConstant() * (_STEdot_max - _STEdot_min))

Notice that for STE_error(Specific Total Energy error) part, the needed throttle(range:0~1) equals to:

delta_throttle = (1 + KI/s) * (STE_err / timeConst) * 1 / (g*(maxClimbRate - minSinkrate))

KI represents for integGain
s represents Laplace operator
THRmax reperents setted value of max throttle ,around 0~1
maxClimbrate represents as velocity keeping at cruise speed the climb rate achieved at THRmax 
minSinkrate represents as velocity keeping at cruise speed the sink rate achieved at THRmin

so as there is a STE_err, we desire compensate  STE_err/timeConst energy per second.
g*(maxClimbRate - minSinkrate) means how much total energy can be offered by (THRmax - THRmin )
Then g*(maxClimbRate - minSinkrate)/(THRmax - THRmin ) roughly estimate the energy generated by one unit throttle.
So I think (STE_err / timeConst) * (THRmax - THRmin ) / (g*(maxClimbRate - minSinkrate)) is a proper estimation for the throttle needed, not the original one without (THRmax - THRmin ) term.

P.S.
In feed forward part for STEdot_dem , 
the feedforward gain equals to (THRmax - THRmin )/[g*(maxClimbRate - minSinkrate)],
which I think it's right. So I think it's good to also multiply (THRmax - THRmin ) for STE_err part, which has a rational physical meaning.

I also draw a simple total energy control block diagram as below, limitation parts are not drawn for simplicity and easy to see. I think it's helpful to debug code.
 
[image]
#### Version
ArduPlane 3.8

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

priseborough, 2017-08-07:
Yes, for consistency the same throttle range should be used for both feed forward and feed-back.

nocternal, 2017-08-08:
Okay, I pull a request #6739 to modify K_STE2Thr

OXINARF, 2017-08-15:
Fixed by #6739

closed 2017-08-15 by OXINARF