ArduPilot/ardupilot #6732: QuadPlane: motors arming by themselves

expntly, 2017-08-06:
#### Issue details
Here's something strange I've noticed when powering on my vehicle.  Depending on whether the radio is on or not when plugging in the batteries, some motors will arm immediately by themselves without me arming anything (safety switch and remote).  This is an X8 configuration.  The motors that arm are the upper ones (on MAIN).  The lower ones (on AUX) keep beeping normally.

Here are the two possible scenarios:

* turn on RC
* plug in batteries to power FC

=> upper motors will arm immediately (rapid beeps followed by silence), lower motors keep beeping slowly (heartbeat) to say they're disarmed

============================

* plug in batteries to power FC
* turn on RC

=> all motors beeping slowly (heartbeat) to say they're disarmed

I re-calibrated the ESCs and the radio to be sure.

#### Version
3.8.0-beta5

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
X8 QuadPlane

#### Hardware type
Pixhawk2.1

magicrub, 2017-08-07:
Please post logs. Also, v3.8 officially released, please check that the problem is still present on that version.

expntly, 2017-08-11:
Aren't logs only generated when you fully arm the vehicle (i.e. safety switch + RC) ?

Yes it still happens with the 3.8 release.

magicrub, 2017-08-12:
There's a param to log while disarmed.

yuxiqun, 2017-08-24:
@expntly Have you sloved your problem that your motors arming themselves? If yes, how do you do it?If no,maybe parameter "RC_TRIM" is not corrected I think.

DavidIngraham, 2018-01-09:
Can this be closed? It seems like a support issue and the OP has not responded

WickedShell, 2018-01-25:
This seems to be a stale support issue, closing.

closed 2018-01-25 by WickedShell