ArduPilot/ardupilot #6735: Copter: Guided pos and vel controller doesn't hold z position, it descends slowly

jverbeke, 2017-08-07:
#### Issue details
I'm using a companion computer to send desired (local) pos and vel commandos to pixhawk.
The pos_and_vel controller is active. I'm sending all zeros, both position and velocity.
This should result in the copter hovering. X and Y are relatively ok. However, the copter descends very slowly (few cm/s) at constant speed and even lands if not countered. Below is a download link for the .bin file. You can see the sonar height decreasing, even though GUID.vZ is zero. When the alt goes below 90cm, my companion computer sends a climb command. But above that altitude, after vz goes back to zero, it slowly descends again. 

I'm not sure what causes this. The inaccuracy of the current position? I'm sending local position commands which are added to the current position to find the new position target. But then I would expect slow random drift, not a constant slow descend.
 
#### Version
ArduCopter 3.4

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
hex

#### Hardware type
pixhawk
#### Logs
https://www.dropbox.com/s/55apkhzw1no9p9u/Guided_pos_and_vel_drift.bin?dl=0

jverbeke, 2017-11-09:
Found the solution myself: the posvel controller doesn't take into account height velocities commanded by companion computer and in addition sonar height to cope with sloped terrain. See pull request: Added changes guided mode posvel controller to incorporate altitude controller #7210

hendjoshsr71, 2021-11-01:
Closing this one as it quite old and many releases ago. With all of the updates to the controllers and guided mode in 4.1, please open a new issue if you have issues in 4.1

closed 2021-11-01 by hendjoshsr71