ArduPilot/ardupilot #6736: Copter: Guided pos and vel controller doesn't react to velocity.z commands

jverbeke, 2017-08-07:
#### Issue details
I'm using a companion computer to send desired (local) pos and vel commandos to pixhawk.
The pos_and_vel controller is active. 
This issue is a separate, but possibly related issue to https://github.com/ArduPilot/ardupilot/issues/6735.

When sending (pos.x,pos.y,pos.z,vel.x,vel.y,vel.z) = (0,0,0,0,0,0.5) to arducopter it should climb with 0.5m/s. However, it does not, except for descending due to ,the possibly related, issue https://github.com/ArduPilot/ardupilot/issues/6735. When sending (pos.x,pos.y,pos.z,vel.x,vel.y,vel.z) = (0,0,-1.5,0,0,0), it responds immediately.
The pos_and_vel controller works by integrating the desired velocity  and adding it to the desired position target in control.guided.cpp. This works for xy_controller but doesn't have effect for the z-controller which is called separately:

> // advance position target using velocity target
> guided_pos_target_cm += guided_vel_target_cms * dt;

When printing out values, it looks like the position error is indeed corrected for the desired velocity, but the next iteration of the update_z_controller, this correction is no longer present. As the correction only happens at pos update rate, most position_target outputs from guided_posvel_control_run() are not compensated for the desired velocity. In addition, when update_z_controller() is called, it calls pos_to_rate_z() which only looks at the position error, not the velocity error as _flags.use_desvel_ff_z is false:

> // add feed forward component
>     if (_flags.use_desvel_ff_z) {
>         _vel_target.z += _vel_desired.z;
>     }

I solved this by changing two items in the code: 

1) Every time new position and velocity target message arrives, the new position is set with pos_control.set_pos_target(position). This function sets _flags.use_desvel_ff_z to false each time and sets _vel_desired.z to zero.  These latter two are commented.
2) Similar to pos controller in guided mode and loiter, but not present in the pos_and_vel controller, the alt target should be compensated by feedforward for the desired climb rate right before calling update_z_controller() using:

> // update altitude target
> set_alt_target_from_climb_rate_ff(_vel_desired.z, _dt, false);

Can anyone give me some feedback on this problem and solution? 
Have  I got the cause right and is this the best solution?

#### Version
ArduCopter 3.4

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
hex
#### Hardware type
pixhawk

amilcarlucas, 2017-10-22:
Can you please open a "github pull request" with your code changes ? I would like to test them out.

jverbeke, 2017-10-24:
Hi Amilcar,

Thanks for your reply.
I have to make the changes to the master version.
I can do that, but I haven’t got the time right now as I’m in the final weeks of my PhD.
I will put it on my TODO list for the beginning of November if that is OK for you.

Kind regards,

Jon

amilcarlucas, 2017-11-09:
The sooner you do it the sooner it will get fixed.

jverbeke, 2017-11-09:
See Pull request: Added changes guided mode posvel controller to incorporate altitude controller #7210

jverbeke, 2017-11-09:
See new pull request: Added changes guided mode posvel controller to incorporate altitude controller #7211

Pokornz, 2021-05-27:
Hi @amilcarlucas ,

This still seems to be an issue (3.5 years later), the guided mode in posvel ignores the vertical component of the velocity target, which makes Ardupilot much less useful for multirotor applications beyond waypoint missions.

I can see that @jverbeke has not been active for quite a while, is there a way to bring this to the attention of some core/active developers?

amilcarlucas, 2021-05-27:
This has probably been recently fixed in git master branch. Can you retest?

amilcarlucas, 2021-05-27:
@Pokornz please continue any discussions on #7211 and not here.

rmackay9, 2021-05-28:
I suspect this was fixed years ago.. certainly the velocity controller accepts climb and descent commands correctly.. I tested this just the other day using the mavproxy "velocity" command.

Pokornz, 2021-05-31:
> I suspect this was fixed years ago.. certainly the velocity controller accepts climb and descent commands correctly.. I tested this just the other day using the mavproxy "velocity" command.

As far as I can see in `ArduCopter/mode_guided.cpp`, it was just fixed in 4.1.0, which is awesome! 
Do you have any rough estimates on when I could expect Copter 4.1.0 to be in stable release?

rmackay9, 2021-05-31:
@Pokornz,

Great, thanks for the feedback.  [4.1.0 is in beta testing](https://discuss.ardupilot.org/t/copter-4-1-0-beta3-has-been-release-for-beta-testing/71489) now but I suspect we are still about 8 weeks from a stable release.  The go-no-go decision is based on how many reports we get and how quickly we can fix them though so it's hard to say how long it will take.

IamPete1, 2021-08-15:
sounds like this is fixed

closed 2021-08-15 by IamPete1