ArduPilot/ardupilot #6747: Plane:error in the calculation of aerodynamic_load_factor

yuxiqun, 2017-08-10:
#### Issue details
In ArduPlane/Attitude.cpp where i found an error about calculation of  aerodynamic_load_factor，which represents the ratio of aircraft lift and itself weight, so the fomular should be " aerodynamic_load_factor = 1.0f / cosf(radians(demanded_roll))" rather than " aerodynamic_load_factor = 1.0f / safe_sqrt(cosf(radians(demanded_roll)))".

#### Version
ArduPlane 3.8.0
#### Platform
Plane
#### Airframe type
eletric flying wing

#### Hardware type
pixhawk

auturgy, 2017-08-12:
This might help a little.  Apologies if it doesn't.
In "aviation speak" yuxiqin is right regarding formula for aerodynamic load factor.  In the code though it may just be a case of the variable name causing confusion.  Aerodynamic stall speed increases with the sqrt of the load factor, which is what the function seeks to describe.

yuxiqun, 2017-08-12:
Thanks for your reply @auturgy firstly, I agree with you that" aerodynamic_load_factor = 1.0f / cosf(radians(demanded_roll)) and  stall_factor = safe_sqrt(aerodynamic_load_factor), but I think there is also another question needing for your attention, please` see down:
>  DataFlash_Class::instance()->Log_Write("TEC2", "TimeUS,KErr,PErr,EDelta,LF", "Qffff",

                                           now,

                                           (double)logging.SKE_error,

                                           (double)logging.SPE_error,

                                           (double)logging.SEB_delta,

                                           (double)load_factor);}
The question is what's the meaning of "LF (the load_factor)",should it is "aerodynamic_load_factor " or " stall_factor ". It's important for me . I found that the actual loiter radius 50m of my plane more than theoretical  30m ,which i considered relevant to the "LF" .For example ,I want my plane loitering with 42 degree roll angle ,the aerodynamic_load_factor should be 1.34( which equals 1/cos(42/180*pi)),but the "LF"in the log is 1.15( which equals sqrt(1.34)),so the "LF" may be "stall_factor ". No matter which meaning it is ,It must influence the flight perfomance of plane.

auturgy, 2017-08-12:
That's what I meant by the function perhaps being confusingly named.  
Load factor (in the normal aviation use of the term) doesn't appear to have been directly calculated/stored in attitude.cpp, so unless there is another function somewhere else in the code that DF is reading from, the logged value will not be the "true" load factor (it will be the "stall_factor").  I'm out of my depth when it comes to the codebase though, so will wait for Tridge and/or Tom to provide some feedback/guidance.  I'm not sure why the code was written the way it is, or where else LF might be calculated, so might do more harm than good fiddling around in there.