ArduPilot/ardupilot #6758: QuadPlane: VTOL motors PWM values fluctuating too much too quickly

expntly, 2017-08-12:
#### Issue details
VTOL motors seem to be sent PWM values that vary at a very high rate, too high and the vehicle experiences stability loss every few seconds or so.

Here's one of the VTOL motors:
[image]

Here's 4 of the VTOL motors:
[image]

I tried to lower the roll and pitch P and I values towards the last segment of that flight, which didn't help.  Vibrations apparently OK.

Any ideas?  Happy to test suggestions.

#### Version
3.8.0 Plane release

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
X8 octaquadplane

#### Hardware type
Pixhawk2.1

#### Logs
DF log: https://www.dropbox.com/s/p5e5ituq7f43dkv/00000039.BIN?dl=0

expntly, 2017-08-13:
So I tried to change to OneShot125 today to get more data points.  I also tried a few values for Q_RC_SPEED, at 150 and 600.  I also did my best to completely rule out vibrations, which now never exceed 6 (for all VibeX, VibeY, VibeZ, no clipping).

I'm still seeing high frequency changes in the signals sent to the ESCs.  Here's 2 of the 8 motors (it gets too noisy with more).

[image]

DF logs with Q_RC_SPEED=150: https://www.dropbox.com/s/1m4zzx5wckwozwa/00000057.BIN?dl=0

DF logs with Q_RC_SPEED=600: https://www.dropbox.com/s/1gvgudwacc7rdln/00000058.BIN?dl=0

expntly, 2017-08-13:
Just to compare, this is what it looks like when it works better (well not the first few seconds):
[image]

The only difference is Q_A_RAT_{RLL,PIT}_{P,I} = 0.5.  But I tried to go back to 0.5 from 0.8 and wasn't able to get a good hover again.  Nothing has changed on the vehicle.  Just different flights on the same day.  It's a bit windy though, but this vehicle has 100kg of thrust for 30kg of mass, so that should be plenty.

DF logs: https://www.dropbox.com/s/0oh8uipm2v63699/00000049.BIN?dl=0

expntly, 2017-08-17:
Any ideas?  Could this be related to PH2.1 even though it looks like a software thing?

expntly, 2017-08-17:
We had another flight where the vehicle was flying normally in QLOITER a few meters above the ground and then started losing altitude -- going full throttle and switching to QSTABILIZE didn't prevent the vehicle from hitting the ground 5s later.

One thing I'm noticing in the logs, is the accelerometers, especially in the Y dimension, registering the same high oscillations.  Could they be the root cause, then causing the FC to send high/low PWM values to the ESCs?

The vehicle starts losing altitude at 140-145 on the graph below:

[image]

expntly, 2017-08-22:
@tridge help please :-)

magicrub, 2017-08-22:
If the accels are going nuts then you either have a prop balance problem, a flexible body frame, or in the case of PH2.1/cube you didn't hard mount the autopilot and it's flopping around.

expntly, 2017-08-23:
Could you please expand on "in the case of PH2.1/cube you didn't hard mount the autopilot and it's flopping around" ?

We use a mount that looks like this today http://imgur.com/bgzBepz

expntly, 2017-08-24:
@proficnc can you please comment on whether this may be a PH2.1 issue and if the way it's mounted is right?

We might try to go back to a regular Pixhawk 1 to test that theory, but lots of rewiring involved.

magicrub, 2017-08-24:
@expntly The PH2.1/cube is internally dampended and any external dampening will make things worse. Please remove your external solution. Note the feature listed at the top:
http://www.proficnc.com/img/cms/Front%20Page%202/pixhawk2%E7%94%A2%E5%93%81%E7%89%B9%E6%80%A7_cube+edison.jpg

expntly, 2017-08-24:
@magicrub OK we'll try but it sounds pretty counter-intuitive that adding anti-vibration + dampening systems will make things worse.  Actually the PH2.1 ships with external foam pads.

Is there precedent for this?  I'd love to understand the logic.

magicrub, 2017-08-24:
Imagine adding 100 layers of "anti-vibration + dampening". The sensors are just flopping around. The real inertial movement of the frame is not sensed. Adding a layer of dampending degrades the frame inertial forces in exchange for remove high-G spikes in the sensor data so that the accels don't clip. Once you add more dampening beyond that you're just degrading degrading with no benefit. In some cases additional dampening is useful but for 99% of the users out there, the dampening material in the Cube has been finely tuned to work for most copters.

proficnc, 2017-08-24:
To start with, this should be in discuss.ardupilot.org as a support question, as this area is to define specific issues.

Can you please try the following two things.

1. With your current setup, please do a flight with high rate logging.
2. Repeat the flight without the additional isolation, with high rate logging.

Please also post pictures of your exact setup.

If your frame has vibration below 100Hz, then low frequency isolation is often needed, but if it doesn't, then as Tom mentioned, it can cause instability due to the additional motion allowed in the system. 

The mounts you show in that photo, cause the system to have a very high centre of gravity, and thus possibly induce horrible roll oscillations 

However, without data, we are all guessing..

So please complete the two suggestions I made above...  and put that info on discuss.ardupilot.org

Please link here. 
I'm closing this, but if we find any issues in the discussion over there, we can raise an additional issue

closed 2017-08-24 by proficnc

expntly, 2017-08-26:
> However, without data, we are all guessing..

@proficnc What do you mean there's **five** .BIN files attached to this thread :-)

Also just to be clear, what LOG_BITMASK value do you recommend for high rate logging?

Sounds good, will post the results on discuss.ardupilot.org.