ArduPilot/ardupilot #6762: UAVCAN: Hangup upon unknown message

olliw42, 2017-08-13:
#### Issue details
It seems that when there is a UAVCAN message on the CAN bus which is not known by ArduPilot, then it hangs up.

#### Version
AC3.6-dev

#### Platform
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
but probably valid for all airframes

#### Airframe type
IMHO irrelevant

#### Hardware type
IMHO irrelevant, but seen on pixracer, AUAV-X2 with DIY can

EShamaev, 2017-09-17:
Please provide more information on what exact message lead to hangup.
I did tests with different messages not processed by AP and none hangs up.
Also please specify the version of AP that you use for that (Commit number)

EShamaev, 2017-11-12:
@olliw42 can the problem be reproduced?

EShamaev, 2018-02-16:
Tested with many messages that are not known to AP and was not reproduced.
If the answer is provided - can be reopened.

closed 2018-02-16 by EShamaev