ArduPilot/ardupilot #6765: Prevent the drone from climbing in Land mode when at least 2 sensors clearly say that it is climbing

lebarsfa, 2017-08-14:
#### Issue details
In some situations (probably EKF down because of high vibrations), the drone climbs in Land mode while baro, GPS and LIDAR-Lite (when applicable) agree between them to say that the drone is climbing while it should not. Therefore, I wonder if there would still be a way to "save" the drone by putting some kind of extra protection such as "do not fly away if at least 2 sensors say that you are climbing while it was asked to land".

Those 2 issues are possibly related in a context of high vibrations : 
_ https://github.com/ArduPilot/ardupilot/issues/4825
_ https://github.com/ArduPilot/ardupilot/issues/3883

#### Version
Copter V3.6-dev (bfd8d151) / Copter V3.5.1 (8ab1b3eb)

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quadrotor / Hexarotor

#### Hardware type
HKPilot32

#### Logs
2 examples of situations where drones were climbing in land mode are available here : http://diydrones.com/forum/topics/land-activation-at-around-1-m-of-altitude-sent-the-quadrotor-at-4 . 
The first was indoor with consistent baro and LIDAR-Lite data, the other was outside with correct baro and GPS at all times (also with LIDAR-Lite when in range). We suppose that high vibrations were the primary cause of the unexpected behavior, but there might still be a way to let the Land succeed. Both lead to a crash for different reasons, but the pilot did not manage to react properly mainly because it was very unexpected that the drone would climb that much in Land mode.

chobitsfan, 2019-03-22:
I have the similar problem, my copters climbing up to dangerous 1200m before it starts to decent
https://discuss.ardupilot.org/t/copter3-5-7-the-drone-will-not-rtl-and-landing-always-fly-up/39978

rmackay9, 2020-01-27:
Copter-4.0 includes a vibration failsafe so I think the core cause is resolved.  We haven't seen any reports of the vibration failsafe occuring but we also haven't seen any "climb away" reports either.

We already have an issue for the rangefinder issue so no need to cover that one again I think.

Anyway, we can always re-open if I've got this wrong

closed 2020-01-27 by rmackay9