ArduPilot/ardupilot #6767: Autotest manifest files broken

jaxxzer, 2017-08-14:
The [firmware-version.txt](http://firmware.us.ardupilot.org/Plane/latest/PX4/firmware-version.txt) files say
`Unknown vehicle (.) (be in a vehicle directory or supply a vehicle type as an argument)`

Also, there used to be a line at the bottom of git-version.txt that said `APMVERSION...` and that is not there anymore. QGC uses these APMVERSION lines to parse the available fws to flash, so some of them (stable Copter or any master branch) are unavailable for download in QGC for now.

http://firmware.us.ardupilot.org/Plane/stable/PX4/ has the APMVERSION line in git-version.txt
where 
http://firmware.us.ardupilot.org/Copter/stable/PX4/ does not, so it would seem the change was introduced sometime between Aug 5 and 14.

peterbarker, 2017-08-15:
Incidentally, these aren't the manifest files; that's manifest.xml :-)

peterbarker, 2017-08-15:
... or is it json? ;-)

jaxxzer, 2017-08-15:
Where is manifest.xml? Also, I don't get the insinuation, sorry. Is it relevant?

jaxxzer, 2017-08-15:
[From what I can tell](https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/build_binaries.sh#L154) we should still be adding the version info in the locations I mentioned.

peterbarker, 2017-08-15:
Quick update.
 - I've fixed the corruption on the server.
 - I've pushed in two commits to try to stop it happening again

More corruption may occur until we get past my commits.

peterbarker, 2017-08-15:
http://firmware.ardupilot.org/manifest.json.gz is supposed to be a one-stop shop for working out what can be downloaded in terms of ArduPilot *stuff*

"insinuation"?  I just forgot it was JSON rather than XML...

We *should* be adding the information to those files.  But we had a few bugs...

jaxxzer, 2017-08-15:
Ok, thank you. This is the first I have seen the manifest.json. cool.

OXINARF, 2017-08-15:
@jaxxzer This should be fixed, have you checked everything is working?

jaxxzer, 2017-08-16:
Looks good.

closed 2017-08-16 by jaxxzer