ArduPilot/ardupilot #6768: Delay sending untrustworthy GPS coordinates

lvale, 2017-08-14:
Connecting a PixHawk (Copter 3.52) to the GCS (in this case MP) as soon as a it boots reveals a strange behaviour where the reported coordinates are false until it gets a good lock of the GPS.

I believe GPS coordinates should only be reported as soon as they are "reliable" to avoid the following:

[image]
[image]

OXINARF, 2017-08-15:
This has been talked some time ago in Gitter and rejected. It makes perfect sense to report what we have to the ground station since it also includes the fix we currently have.

In my opinion this should be closed and GCS should handle it.

@WickedShell what's your take on this?

WickedShell, 2017-08-15:
My opinion is that the GCS should show whatever, I think this is the GLOBAL_POSITION_INT drifting which is valid but I need to see a tlog from @lvale which we talked about on mumble.

WickedShell, 2017-09-15:
Closing as a GCS side issue (never got a log to look at, but sending our best estimate of GLOBAL_POSITION_INT and relative motion on it is still absolutely valid IMO)

closed 2017-09-15 by WickedShell