ArduPilot/ardupilot #6770: Ability to land or terminate flight on GUIDED command stream loss

comran, 2017-08-15:
#### Issue details
When Guided mode connection is lost (whether connection to the companion computer is lost or the companion computer Dronekit code crashes), the copter will only go into RTL without any parameter to change this to LAND or potentially flight termination. For my program using Dronekit where there are a swarm of quadcopters flying together, landing would be a lot safer than RTL because RTL has the potential to collide with other quadcopters or potentially obstacles that I am avoiding with my program that feeds velocities to the copter over Dronekit/Mavlink.

#### Version
3.5.1

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
quad

#### Hardware type
Pixhawk

srikarg89, 2019-05-12:
Have you made any progress with this? I'm facing the same problem but for Plane.

peterbarker, 2021-07-30:
Copter's had FS_GCS_ENABLED_ALWAYS_LAND for quite some time now.

closed 2021-07-30 by peterbarker