ArduPilot/ardupilot #6773: Plane: better management of the "soaring mode" aimed at endurance

robustini, 2017-08-16:
#### Issue details
The "soaring" mode is very useful if you use Ardupilot with a glider, but unfortunately it lacks an important feature for those who want to make automatic missions using thermal and dynamic with a normal plane.
During a endurance tests in "AUTO" flight mode with my MiniTalon i often found thermal, but unfortunately the autopilot tried to keep the plane at the set height of the mission.
The 5 waypoints were all placed at 70 meters, and often the plane proceeded without propelling for the presence of thermal.
In that condition the autopilot forced the plane to stay at 70 meters, thus losing the chance to exploit such heat to heighten.
What I would like to implement is a rather simple concept but functional in these cases, for example by setting existing paramaters without adding another.
In detail, an example:

SOAR_ENABLE,2
SOAR_ALT_MIN,70
SOAR_ALT_MAX,350

During an automatic mission with fixed waypoint at 80 meters if the plane gets thermal condition is not bound to stay at 80 meters but can climb up to 350, and once reached that height the TECS will prevent it from going up further.
If the plane tends to fall below 80 meters, the autopilot will obviously give the engine so as to keep it above that height, as it already does.
With the "SOAR" parameters this thing can't be done at this time.

#### Version
Plane V3.8.0

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

#### Airframe type
MiniTalon and other...

#### Hardware type
Pixhawk

#### Logs

samuelctabor, 2017-08-17:
Hi Marco,

Good idea. 
I think that with SOAR_ENABLE=0 and TECS_SPDWEIGHT=2 the TECS will just reduce throttle (possibly to zero, I'm not sure) if it is lifted above the waypoint altitude by a thermal.
So perhaps the required functionality is to set TECS_SPDWEIGHT=1 while this happens.
Would that achieve the effect you're looking for?

magicrub, 2017-08-17:
@samuelctabor what he's saying is if a given way point is x, don't limit yourself to x. Allow it to go all the way up to y during normal operation. Basically, lower bound is enfoced but upper altitude bound is encouraged to exceed where possible

robustini, 2017-08-18:
Thanks Tom for a better explaination.
And of course there must be a point z where the plane beyond can not go up (max altitude).
@samuelctabor, with what you tell me to do I don't get the desired result.

robustini, 2017-08-18:
This is the TLOG of my flight with my MiniTalon, over three hours in full automatic mission.
If you analyze it you can see that there was sometimes a wind holding the plane it in the air, it's awesome, the wind at the same speed of the Cruise speed.
In some conditions the motor is switched off because it finds thermal, but the autopilot binds it to the set height, thus losing efficiency...

https://drive.google.com/file/d/0B3HEcrNCOga9SVBVd1NlTllCUlk/view?usp=sharing

samuelctabor, 2017-09-03:
Hey Marco, having a look at this now: are you sure that's the right log? I can't see any events where the motor was switched off in AUTO mode ( it did go down to around 10% at one point).

Cheers, S

robustini, 2017-09-03:
Hey Samuel, i'm not sure about that log, but I'm sure the code behaves that way, force the plane to stay at the "waypoint or CRUISE height", even if there are thermal.

magicrub, 2017-09-04:
It feels like we need to add something like this to [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_TECS/AP_TECS.cpp#L783):
```
else if (_soaring_controller.is_active() == 2) {
    // if soaring, focus on airspeed and allow sloppy height management
    SKE_weighting = 2.0f;
}
```

and [this](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_TECS/AP_TECS.cpp#L970) needs something like:
```
if (_soaring_controller.is_active() == 2 &&
    _soaring_controller.isInThermal() &&
    _flags.underspeed == false &&
    alt < _soaring_controller.alt_max &&
    alt < mission.alt)) {
    // if soaring and we're in a thermal, ride it upwards and don't pitch down
    _pitch_min = 0;
} else if (_pitch_min >= 0) {
...
```

robustini, 2017-09-04:
Yes Tom, something like this!

magicrub, 2017-09-05:
@samuelctabor  What is a good way to simulate thermals for SITL testing?

magicrub, 2017-09-05:
ahh, I see. You were using X-Plane. I'll see about adding some magical ones to JSBsim for easier crude testing

samuelctabor, 2017-09-10:
I've had a go at implementing this - https://github.com/samuelctabor/ardupilot/tree/soaringForPoweredAC
It needs some more testing but you're welcome to give it a go. It simply sets the SKE weighting to 2.0 if SOAR_ENABLE==2 and alt< SOAR_ALT_MAX. I haven't modified the pitch limits. My test aircraft is not behaving well in simulation so I need to do some debugging on that.
For testing I usually use JSBSim with some 'magic' thermals - see branchs thermalsJSB or soaringForPoweredAC-dev.

Cheers, Sam

closed 2020-11-30 by robustini