ArduPilot/ardupilot #6785: Plane: Maxsonar type define conflict

ranqingfa, 2017-08-17:
Issue details
In Plane firmware 3.8 the rangefinder type number(parameter) of Maxsonar is changed from 4 to 2（MaxsonarI2C），but in the source code of AP_RANGEFINDER_MAXSONARI2CXL.h（Line 9）there is a define: #define AP_RANGEFINDER_MAXSONARI2CXL    4, it's a conflict! The sonarrange data displaying on GCS is always updated delayed for several senconds. I've changed the define value to 2, sonar distance data is updated normal now. But still there is another problem: if I set RNGFND_LANDING to 1, ardupliot always show "Bad LiDAR Health" now, if set to 0, there is not. Im still working on it, someone could help or give some tips of what codes should I looking into.

Version
Plane 3.8 stable 

Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

Airframe type
Fixed Wing

#### Hardware type
Pixhawk

#### Logs

OXINARF, 2017-08-17:
There is no conflict. That macro isn't used anywhere (you are welcome to do a PR to remove it), it doesn't make any difference changing it to 2 or whatever value.

Regarding the other problem, please post in the forum. As you know, this isn't the place for user support. But looking at your graph it seems pretty clear that something is wrong with your rangefinder, start by checking the cable.

closed 2017-08-17 by OXINARF

ranqingfa, 2017-08-17:
This sonar works well in firmware 3.7.1!I just have test it today!

OXINARF, 2017-08-21:
Post logs in the forum.