ArduPilot/ardupilot #6796: SafeRTL suggestion:

AndKe, 2017-08-20:
For some missions, close to obstacles, it would be great to have a SafeRTL paramteter to increase Altitude by "x"  (tracking back home, adding some altitude to all recorded points.)

proficnc, 2017-08-20:
As safe RTL is 3D, adding any offset would be inherently dangerous. The current method allows for the possibility that the vehicle may have traveled under a tree, bridge etc on the way out...

rmackay9, 2017-08-21:
Maybe the more general request here though is to do object avoidance during RTL (and other modes) which is actually already on the to-do list.  I've updated the title in any case.

lvale, 2017-08-21:
Adding a offset to a safe path breaks the purpose of it. Can only envision this with the aid of SLAM and Object Avoidance.

AndKe, 2017-08-22:
The intension bedding my request is high voltage line inspection. While a lateral deviation from the line may hit the trees (often higher than 66 and 132kV line) 
Holding altitude is no good, as we fly 20cm above the wires to photograph directly into chain links to assess water. Moving around the masts. Much too high risk to reproduce by replay.

An RTL we use for safety, as is today, does not attempt to follow path, just adds a preset altitude, which is rather high climb, to ensure clearance over trees.

An parameter for altitude offset, would ensure safe return when airspace above is safe, or below (when doing a bridge inspection. RTL could offset by a negative number, to get a bit lower)