ArduPilot/ardupilot #6798: Spline WP + LOITER_UNLIM combination causes Floating point exception for ArduCopter.

MCharikov, 2017-08-21:
#### Issue details

Have a mission which crashes ArduCopter SITL (didn't check on real hardware) with Floating point exception. 

I have mission with two spline waypoints with equal lat\lng\alt and LOITER_UNLIM cmd between them (see attach). Idea was to enter LOITER_UNLIM state, wait for continue command and after 5 seconds continue flight. Look at 2,3 and 4 waypoints in mission. Copter successfully enters LOITER_UNLIM (WP3) and after i set WP4 as next WP (in Mission Planner for example) - SITL crashes. 

For some reason i used WAYPOINT_SPLINE with delay as WP4 instead of using LOITER_TIME. With LOITER_TIME mission works well. 

[mission_broke_waypoints.txt](https://github.com/ArduPilot/ardupilot/files/1238491/mission_broke_waypoints.txt)

#### Version
3.5.1

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quad. Tests were made on SITL only. 

#### Hardware type
SITL. Didn't check on real hardware. 

#### Logs
_Please provide a link to any relevant logs that show the issue_

khancyr, 2017-08-21:
Hello, 

Thanks for the report ! I confirm the bug on copter master too (present at https://github.com/ArduPilot/ardupilot/commit/b0c7766197aaa19236a42dd6f57221391c759899)
It comes from https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_WPNav/AC_WPNav.cpp#L1089 with target_vel = (0;0;0) so we get a divide by 0 !

rmackay9, 2017-08-21:
Great, thanks for the report.  I'll add it to the list to sort out for Copter-3.5.3.

rmackay9, 2017-11-23:
I'm having a hard time recreating this problem with master.  I don't think we've fixed it but I just can't make it crash..

rmackay9, 2017-11-23:
Here's a fix from @lthall and I.  It's similar in some ways to @khancyr's solution although we've decided to just mark the vehicle as having reached the destination because we think this issue can only occur if the origin and destination are the same point.

closed 2017-11-23 by rmackay9