ArduPilot/ardupilot #6802: Copter: CONDITION_YAW command doesn't work if it runs directly after TAKEOFF in GUIDED mode

MCharikov, 2017-08-21:
#### Issue details

if we run commands from console, UAV will not turn after that:

1. arm throttle
2. mode guided
3. takeoff 10
4. setyaw 90 5 0

But if we add position change between takeoff and condition_yaw, UAV will turn perfect:

1. arm throttle
2. mode guided
2. takeoff 10
3. position 1 0 0
4. setyaw 90 5 0

#### Version
3.5.1

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
SITL, quad, hexa

#### Hardware type
SITL, pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-08-21:
This seems to work as intended because we don't support setting yaw angle during takeoff - this applies to missions too. Also until you ask Guided for another mode, you are in takeoff mode.

I've changed the title to make this a feature request.

MCharikov, 2017-08-22:
I see. It seems that autopilot need to detect that it finished takeoff by itself.

IamPete1, 2021-08-30:
This does work if you wait for a while after the takeoff, I suspect the fix is to stop accepting yaw commands if they wont have any effect.

rmackay9, 2021-08-31:
I agree that 1/2 the problem is resolved now in that the companion computer doesn't need to detect the takeoff has completed.  @khancyr resolved this portion of the issue a couple of years ago.  The other 1/2 is still outstanding though, the vehicle's heading cannot be controlled while it is still climbing.