ArduPilot/ardupilot #6809: cannot disable compass on pixhawk 2 with plane 3.8.

geofrancis, 2017-08-22:
I have disabled all but one of the compass sensors but due to the proximity of one of the compass's to a camera but it keeps sending "bad compass health" messages. the only way i can get the messages to stop is to disable compass use alltogether but then i loose wind speed estimations.

WickedShell, 2017-08-22:
Bad compass health indicates a failure to read from your primary compass within the last half second. What is your primary compass set to (and what are the ID's).

Was this not a problem on plane 3.7 or is this the first time that the PH2 was setup?

geofrancis, 2017-08-23:
its its first build, but i think you might be right, mag field disturbance would throw up a "check mag field" error not bad compass health.

OXINARF, 2017-08-23:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

closed 2017-08-23 by OXINARF

geofrancis, 2017-08-27:
Sorry I thought it was a bug but it was just me having a brain fart.

geofrancis, 2017-08-27:
Lidar was causing issues, unplugged that and the errors went away