ArduPilot/ardupilot #6814: ossible Serious Safety flaw related to Arm/Disarm in Octo frame

VTOLDavid, 2017-08-23:
Hello everyone, I have been configuring a new Octo using a DJI s1000 as a base and Pixhawk. Everything was going well until I have realised that when the copter is disarmed (the safety switch has not been pushed yet), not every Motor Output Pin is sending the same signal. It seems to me that only outputs from 1 to 4, both inclusive, are put to LOW, which I think is the correct signal, but motors from 5 to 8 are sending a PWM of 1120 (MOT_PWM_MIN value in my case). I have realised of this because the DJI ESC beep when they are not reading a valid PWM value between 1120 and 1940, so I expected all of them beeping then disarmed. I think this could be dangerous in some cases and all motor outputs should be put LOW or at least choose the value in Mission Planner.

This behaviour doesn't apear when I disable the safety switch and only left the stick control to arm.

I am missing something here? 
Thanks.

PS. Even if i leave the safety switch unpushed (light blinking) I can arm the copter with my rx stick (only the motors wich were seeing the PWM start to spin), which is a major safety problem I think.

#### Version
3.4.6 & 3.5.2

#### Platform

[  ] Copter

#### Airframe type
octa

#### Hardware type
pixhawk

[1 01-01-2000 1-37-04.zip](https://github.com/ArduPilot/ardupilot/files/1246576/1.01-01-2000.1-37-04.zip)

VTOLDavid, 2017-08-24:
It seems that the problem was BRD_SAFETY_MASK set to 16368 (Ok for a quadcopter) instead of
BRD_SAFETY_MASK 16128 (Ok for a octo). So BRD_SAFETY_MASK default value for octocopter is not correct.

OXINARF, 2017-08-24:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

The default is setup based on the motor outputs, so my best guess is that you loaded a parameter from a different vehicle and it saved the parameter to the wrong value - the default is then ignored. If you can reproduce this right after a parameter reset, we can re-look at this.

closed 2017-08-24 by OXINARF

VTOLDavid, 2017-08-30:
No, the problem is that the default for a octo is the same that for a quad for the parameter BRD_SAFETY_MASK

OXINARF, 2017-09-01:
@VTOLDavid The default is not the same - as I said, it depends on the motor outputs. I just tested this both in 3.5 and master and it worked. I even tried changing the motor1 output to pin 8 in a quad configuration and the default was also correct.