ArduPilot/ardupilot #6830: Local position mavlink telemetry in master and Copter-3.5 uses Home instead of Origin

sharvashish, 2017-08-25:
#### Issue details
https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L1420

Local Position telemetry mavlink in both 'master' and 'Copter-3.5' branch uses get_relative_position_NED_home() instead of get_relative_position_NED_origin().

Because of this, there is a discontinuity in local position telemetry data, whenever the vehicle is armed away from the point of origin.

Is this something which is desired? As I followed Copter-3.4 and I found that it uses get_relative_position_NED() which is similar to get_relative_position_NED_origin() in master and Copter-3.5 branch.

#### Version
Latest branch master and Copter-3.5
#### Platform
Copter

#### Airframe type
SITL - quad

khancyr, 2017-08-31:
you seem right to me. the fix will be easy. @rmackay9 what do you think ?

OXINARF, 2017-09-01:
Yes, as I've answered in Gitter, I agree this should use the _origin_ version and not _home_. This was probably a wrong change when we changed to have _home_ and _origin_ methods.

WickedShell, 2017-09-01:
I disagree. EKF origin can drift (or has been known to) home is where you actually armed the aircraft and relative to it will be consistent through out a flight.

OXINARF, 2017-09-02:
@WickedShell Some points:

- EKF origin drift is only in altitude, not lat/long
- EKF origin drift is now off by default in master
- EKF origin drift actually attempted to correct it to the truth altitude using GPS
- in master you can now correct the EKF origin or the altitude reported by EKF or none (the default)
- origin can't be changed after it is set
- home is definitely not guaranteed to be consistent during a flight, you can change home whenever you want. Tower even has a Return to Me feature that constantly updates your home location to your position.
- LOCAL_POSITION should have a fixed reference for the entire time the vehicle is turned on and that can't be home (as, at least, it changes on every arm) but only EKF origin

sharvashish, 2017-09-02:
Yes. I second @OXINARF. The fact that mavlink local position uses home, and IIUC position controller uses EKF origin to calculate local position, the copter behaves incorrectly if I send position setpoints via mavlink relative to current position in guided mode.

Say i want the copter to move 10m in north, I get current local position and add 10m to x local position, and then send the final local position setpoint. But the copter does not go to desired location because of the aforementioned issue.

amilcarlucas, 2017-09-21:
@rmackay9 you mention something about NE coordinates being fixed in 3.5.3.
Is this merged in 3.5.3 ?

rmackay9, 2020-09-01:
This is now resolved in master, thanks!

closed 2020-09-01 by rmackay9