ArduPilot/ardupilot #6846: Feature request: use aux pin relay out for led and buzzer

geofrancis, 2017-08-27:
The way the old APM would drive it's external leds and its buzzer was very useful, since it was just a ping going high/low you could make it run any kind or colour of lights you want using an external driver. 

The pix hawk has relay pins that very few people use so why not allow them to be used for driving buzzers and leds like an APM.

sashan13, 2017-09-04:
Totally agree. I guess it may also be possible to reconfigure ADC pins as output?

007trains, 2018-10-24:
Has anyone looked into this? I'd like to be able to trigger a relay in on my plane. Similar to the ch7, ch8 aux funtions in Arducopter

peterbarker, 2018-10-24:
@007trains You should already be able to use those channels for relays on Plane.  Looks for the RELAY* parameters.

My PR to clean up the buzzer sequence code enables use of the buzzer (as opposed to the "tonealarm" you usually hear on a PixHawk) on all platforms: https://github.com/ArduPilot/ardupilot/pull/9534

007trains, 2018-10-25:
@peterbarker Ah i see what you mean, i guess my issue is more like #7071, i have setup the servox_function to relay but cannot set RCn_OPTION as it does not appear in the parameters list

peterbarker, 2021-07-30:
We allow setting this stuff through auxillary pins on stm32 hardware in the hwdef files.

closed 2021-07-30 by peterbarker