ArduPilot/ardupilot #6850: RTOS, FuncCallBacks, AP_HAL_MAIN(), C wrapper, Scheduler, Nuttx, Code Flow, Build Own Minimal autopilot

rajucoolsuraj, 2017-08-27:
#### Issue details
_Please describe the problem, or desired feature_
The ardupilot is a multi-threaded based application. I would like to know the Code Flow or How the RTOS is scheduling the multiple mains -- AP_HAL_MAIN() wrapper to call setup() and loop() in the callbacks.
Please assist me in learning the software architecture of the ardupilot as a multithreaded application.
How drivers are classified as timer, IO? 
other layers of HAL and finally the RTOS - Nuttx scheduling the tasks and threads on Pixhawk platform.
Please assist me in filling the gaps of the insights gained in understanding the code base.

#### Version
_What version was the issue encountered with_
Ardupilot latest version/release -- 
#### Platform
[  ] All
[  ] AntennaTracker
[yes  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
Copter is what I'm building for.
#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
QUAD 'X'
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk V1
#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-08-28:
This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot). Also, a good starting point is reading our developer wiki (http://ardupilot.org/dev/index.html).

closed 2017-08-28 by OXINARF

rajucoolsuraj, 2017-08-29:
Sure,
thank you