ArduPilot/ardupilot #6854: Copter: compilation failed  ,please help me!!

zhangwunian, 2017-08-28:
#### Issue details

when I run my code......
make px4-v2 
// BUILDROOT=/f/git/ardupilot/Build.ArduCopter HAL_BOARD=HAL_BOARD_PX4 HAL_BOARD_SUBTYPE= TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="0f08207b" -I/f/git/ardupilot/libraries/AP_Common/missing -DMAVLINK_PROTOCOL_VERSION=2 -DNUTTX_GIT_VERSION="c626db4d" -DPX4_GIT_VERSION="7a22d4e4" -DUAVCAN=1 -D__STDC_FORMAT_MACROS -DHAVE_STD_NULLPTR_T=0 -DHAVE_ENDIAN_H=0 -DHAVE_BYTESWAP_H=0 -DHAVE_OCLOEXEC=0 -I/f/git/ardupilot/Build.ArduCopter/libraries/GCS_MAVLink/include/mavlink -I/f/git/ardupilot/modules/uavcan/libuavcan/include -I/f/git/ardupilot/modules/uavcan/libuavcan/include/dsdlc_generated
Checking modules
echo "Generating MAVLink headers..."
Generating MAVLink headers...
#goto mavlink module directory and run the most recent generator script
echo "Generating C code using mavgen.py located at" /f/git/ardupilot/modules/mavlink/
Generating C code using mavgen.py located at /f/git/ardupilot/modules/mavlink/
PYTHONPATH=/f/git/ardupilot/modules/mavlink/ python /f/git/ardupilot/modules/mavlink//pymavlink/tools/mavgen.py --lang=C --wire-protocol=2.0 --output=/f/git/ardupilot/Build.ArduCopter/libraries/GCS_MAVLink/include/mavlink/v2.0 /f/git/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml; if [ $? -le 0 -o $? -gt 128 ]; then echo "mavgen: success"; exit 0; else echo "mavgen: failed"; exit 1; fi
Traceback (most recent call last):
  File "f:/git/ardupilot/modules/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "f:\git\ardupilot\modules\mavlink\pymavlink\generator\mavgen.py", line 12, in <module>
    from future import standard_library
ImportError: No module named future
mavgen: failed
make: *** [/f/git/ardupilot/Build.ArduCopter/libraries/GCS_MAVLink/include/mavlink/v2.0/ardupilotmega/mavlink.h] Error 1

#### Version
copter_3.5.2

#### Platform

[Copter

#### Airframe type
hex

#### Hardware type
pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

khancyr, 2017-08-28:
You forget to install future library (sudo pip2 install future) , please follow wiki instructions to set up you environement : http://ardupilot.org/dev/docs/building-the-code.html

closed 2017-08-28 by khancyr

zhangwunian, 2017-08-28:
@khancyr  how to install future library ,did you have some  documents to introduce this? My environement everything was normal ,In the previous version