ArduPilot/ardupilot #6863: Flight not logged. ARMING_CHECK for logging not enforced.

AndKe, 2017-08-29:
AC3.5.2 on Pixhawk

Recently upgraded from AC3.4.6 - that worked fine.
Usually, those machines get 16-20 flights a day, each flight take a set of photos (timestamped in exif)

Today, I got one flight worth of photos that did not had any log for that period of time.
Previous photos were found in log 00000360.BIN  , (then there is a lot of photos for that missing flight)  and the following photos were found in 00000361.BIN.

It's clear that there was a flight between the 360's landing and 361, the GPS time in between also confirms that.   (and following logs were fine)

ARMIMNG_CHECK=1  , so I would not expect to get airborne with a logging error.

Somehow, it's still possible to takeoff with no logging.

Georacer, 2017-08-29:
I recently had the same problem in ArduPlane 3.7.1 and I have just posted about it today in the forums: https://discuss.ardupilot.org/t/logging-check-is-not-enforced/20521

OXINARF, 2017-08-29:
@AndKe Was there a reboot between these flights?

AndKe, 2017-08-30:
@OXINARF yes, due to battery change. I never use ground power between flights, only before flight, on fixed wing with more complex payload or airspace.

AndKe, 2017-08-30:
@OXINARF I would like to do some testing to check how often this occurs on 3.5.2 vs 3.4.6
Would you say power on, arm, disarm, power off (then repeat)  is a good way to reproduce it ?

AndKe, 2017-08-30:
completed test, 10 logs spawned as expected.  
BTW: I always mount the SD cards on computer as read-only.  
And as @Georacer mentioned, I did see something like this on AP3.7.1 as well, once.

Georacer, 2017-08-31:
@AndKe Without thinking too much on it, I re-formatted my uSD before (successfully) trying to capture another log, which ruined the test of uSD corruption.
Did you re-format yours?

AndKe, 2017-08-31:
@Georacer No, the missing log was in between properly logged flights. I did not format, or in any way modify on another system (mounting it as read-only)  - the test yesterday created 10 files fine.

I too had a corrupted filesystem (a year ago) then nothing would ever be logged to that card - formatting solved it. I do not consider it to be the same issue.

What I miss, is a detection of the condition when a file is not created.

peterbarker, 2017-09-14:
@AndKe https://github.com/ArduPilot/ardupilot/pull/6936 fixed a bug where we were not correctly reporting a failure in DataFlash_File.

I'm also in the middle of negotiating fixes to a race condition in DataFlash_File which could trigger such a failure: https://github.com/ArduPilot/ardupilot/pull/6944

AndKe, 2017-09-14:
@peterbarker Well done. Thanks for looking into this issue, it's a very important and elusive one :)

AndKe, 2017-09-28:
Did the #6944 fix made it to 3.5.3 release ?  
If so: was it supposed to affect mid-air log-stop ?

I ask because my previous reports are based on Pixhawk 1.
Now, Today I few some shakedown flights with a big X8 with pixhawk 2.1 and AC 3.5.3  
The log ended after ~1000seconds of a 42minute - it just ends midair.

I am starting to be desperate for a way to avoid this issue.

Config: (defaults)
LOG_BACKEND_TYPE,1
LOG_BITMASK,176126
LOG_DISARMED,0
LOG_FILE_BUFSIZE,16
LOG_FILE_DSRMROT,0
LOG_REPLAY,0

peterbarker, 2017-12-19:
More fixes are in the latest 3.5, too.

Should we close this now and open again when something more happens?

AndKe, 2017-12-19:
...yes, can do, I plan to some 10 test flight before next commercial flight with that UAV.  - will update it to current release before testing then.

closed 2017-12-19 by AndKe