ArduPilot/ardupilot #6875: Rover: Loiter command support/documentation

oysstu, 2017-09-01:
#### Issue details
In the [docs](http://ardupilot.org/planner/docs/common-mavlink-mission-command-messages-mav_cmd.html#mav-cmd-nav-loiter-unlim) the loiter commands are marked as supported for rover. The commands are however treated as a regular waypoints. In the [rover supported commands list](http://ardupilot.org/rover/docs/common-mavlink-mission-command-messages-mav_cmd.html#common-mavlink-mission-command-messages-mav-cmd-commands-supported-by-rover), the loiter commands are not listed.

I suggest either removing rover support for loiter in the first docs page I linked, or implement proper support for a loiter command. 

#### Version
HEAD (c046824)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ x ] Rover
[  ] Submarine

#### Hardware type
Software in the loop, tested on Pixhawk 2 previously

gmorph, 2017-09-02:
Loiter is implemented already.  Its not in a formal release yet.

rmackay9, 2018-02-01:
I think I'll close this issue now that Rover-3.2 has gone out which includes support for the Loiter mission commands.  It's true that it's basically handled just like a regular waypoint but I think that's the correct way to handle it - it's consistent with what Copter does so.. seems correct to me.

closed 2018-02-01 by rmackay9