ArduPilot/ardupilot #6892: COPTER - add WPNAV_JERK parameter

sashan13, 2017-09-04:
Issue details

Right now we have WPNAV_LOIT_JERK parameter. It smooths loiter flight nicely without desired attitude changing by 60 degrees in one step.

However, in auto flight, there's no such limit. Copter jerks around pitch axis very rapidly upon reaching the waypoint, which doesn't look nice and is potentially dangerous. So, what about adding similar WPNAV_JERK  parameter to Auto?  

Version

3.5.2

Platform

[ ] All
[ ] AntennaTracker
[X] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

Airframe type

4kg Quadcopter

Hardware type

Pixhack by CUAV

Logs

No logs

rmackay9, 2017-09-04:
I think it would be best to have logs because I suspect the issue is not the lack of a jerk parameter.  The way that Loiter and waypoint navigation work is quite different and it shouldn't really be necessary to have a jerk limit during waypoint navigation.

sashan13, 2017-09-04:
I will try to obtain logs. But I'm talking about the fact that copter flies in auto in quite sharp way, changing attitude rapidly. Even if I deflect sticks all the way I can't make it fly that way in loiter.

rmackay9, 2017-09-04:
@sashan13, it really should be more gentle than Loiter - so let's see those logs.

sashan13, 2017-09-04:
I've changed Loiter settings quite a lot - it feels much more responsive now. Here are what I changed:
WPNAV_LOIT_JERK,4000
WPNAV_LOIT_MAXA,500
WPNAV_LOIT_MINA,200
WPNAV_LOIT_SPEED,1400
I will post logs when I have them.

sashan13, 2017-09-05:
Here, I have logs. It was a small mapping test flight made from Tower app (mission planned in Tower) with manual takeoff and landing. 

The copter flies in Loiter very well. It flies in Auto quite decently but I really don't like the way it changes flight direction rapidly at each waypoint. Isn't there a way to make it turn smoothly? 

The copter is also somewhat unbalanced but flies OK. I've used quadV mode because its mixing table is the closest to what I've calculated.

[00000010.zip](https://github.com/ArduPilot/ardupilot/files/1277490/00000010.zip)

peterbarker, 2021-07-30:
We have one now.

closed 2021-07-30 by peterbarker