ArduPilot/ardupilot #6897: Custom Status LED not working in Copter 3.5

d-skinner, 2017-09-05:
#### Issue details
I use a custom status LED on the back of my copters that hooks into the I2C1 line and listens in to the messages being sent to the Here GPS status LEDs. The custom LED uses an ATtiny85 to emulate the Toshiba LED driver.

The issue is; updating from ArduCopter 3.4.5 to ArduCopter 3.5.2 makes the custom LED stop working. Absolutely no hardware was changed between the update. Mission Planner was used to jump between the two different versions of ArduCopter to reproduce the problem.

It seems that there has been some change to the status LED code (AP_Notify???) between AC3.4 and AC3.5?

The ATtiny is a little picky with the I2C timing so, my guess is that it has something to do with that as the Here GPS is working perfectly in both AC3.4 and AC3.5 (ie. I2C messages are getting through).

#### Platform & Version
ArduCopter 3.5

#### Hardware type
- Pixhawk2.1 Black Cube
- Here GPS
- RFD900+

d-skinner, 2017-09-05:
@tridge @rmackay9

OXINARF, 2017-09-06:
This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

closed 2017-09-06 by OXINARF

d-skinner, 2017-09-06:
For what it's worth, Tridge helped me figure this out. The issue was caused by the updated Toshiba LED driver that now uses the 4-byte addressing protocol instead of the old 6-byte protocol.

My LED was only looking for a 6-byte protocol.