ArduPilot/ardupilot #6899: modified example/Printf with time.h and signal.h not compiling

OscarJGV26, 2017-09-05:
Hi,

I am trying to compile a modified example (example/Printf) with the libraries <time.h> and <signal.h> on it but when I execute ./waf --target examples/Printf --upload it gives me the following error. 

root@beaglebone:/home/debian/ardupilot# ./waf --target examples/Printf --upload
Waf: Entering directory `/home/debian/ardupilot/build/blue'
[817/817] Linking build/blue/examples/Printf
libraries/AP_HAL/examples/Printf/Printf.cpp.1.o: In function `main':
Printf.cpp:(.text.startup.main+0xb4): undefined reference to `timer_create'
Printf.cpp:(.text.startup.main+0xde): undefined reference to `timer_settime'
collect2: error: ld returned 1 exit status

Waf: Leaving directory `/home/debian/ardupilot/build/blue'
Build failed
 -> task in 'examples/Printf' failed (exit status 1):
        {task 3059501696L: cxxprogram Printf.cpp.1.o -> Printf}
['/usr/bin/arm-linux-gnueabihf-g++', '-Wl,--gc-sections', '-pthread', 'libraries/AP_HAL/examples/Printf/Printf.cpp.1.o', '-o/home/debian/ardupilot/build/blue/examples/Printf', '-Wl,-Bstatic', '-Llib', '-lap', '-Wl,-Bdynamic', '-lm', '-ldl']

I assume this is because the compiler is not using  -lrt to compile. How can I get it to use it? Removing all the hal libraries I am able to compile this code by including #747 
#include <stdlib.h>
#include <unistd.h>
#include <stdio.h>

and replacing hal.console->printf simply with printf 

Could someone help me with this please?

Here is the fairly simply code. 

#include <AP_Common/AP_Common.h>
#include <AP_HAL/AP_HAL.h>
#include <stdlib.h>
#include <unistd.h>
#include <stdio.h>
#include <signal.h>
#include <time.h>
#define CLOCKID CLOCK_REALTIME
#define SIG SIGRTMIN

timer_t timerid;
struct sigevent sev;
struct itimerspec its;
struct timespec real_time_clock;
long long freq_nanosecs;
sigset_t mask;
struct sigaction sa;
timer_t timerid2;
struct sigevent sev2;
struct itimerspec its2;
sigset_t mask2;
struct sigaction sa2;
double rt_clock=0;
double timer=0;
double real_period=0;
int print=0;

static void handler(int sig, siginfo_t *si, void *uc)
{
    clock_gettime(CLOCK_REALTIME,&real_time_clock);
    rt_clock=real_time_clock.tv_sec+(double)real_time_clock.tv_nsec/1000000000.0f;
    real_period=rt_clock-timer;
    timer=rt_clock;
    print=1;
}

const AP_HAL::HAL& hal = AP_HAL::get_HAL();

int main(int argc, char *argv[])
{
   hal.console->printf("Starting Printf test\n");
   sa.sa_flags = SA_SIGINFO;
   sa.sa_sigaction = handler;
   sigemptyset(&sa.sa_mask);
   sigaction(SIG, &sa, NULL);
   sigemptyset(&mask);
   sigaddset(&mask, SIG);
   sigprocmask(SIG_SETMASK, &mask, NULL);
   sev.sigev_notify = SIGEV_SIGNAL;
   sev.sigev_signo = SIG;
   sev.sigev_value.sival_ptr = &timerid;
   timer_create(CLOCKID, &sev, &timerid);
   freq_nanosecs = 1000000000;
   its.it_value.tv_sec = freq_nanosecs / 1000000000;
   its.it_value.tv_nsec = freq_nanosecs % 1000000000;
   its.it_interval.tv_sec = its.it_value.tv_sec;
   its.it_interval.tv_nsec = its.it_value.tv_nsec;
   timer_settime(timerid, 0, &its, NULL);
   sigprocmask(SIG_UNBLOCK, &mask, NULL);
   while(1){
    if(print==1){
        hal.console->printf("Hi... \n");
    }
    }
}
#### Hardware type
Linux - Beaglebone Blue

OscarJGV26, 2017-09-05:
I am going to answer my own question just for those who have problems with librt as well... 

Checking the file /Tools/ardupilotwaf/cxx_checks.py, the function check_librt its looking for the fragment 

fragment='''
        #include <time.h>
        int main(){
        clock_gettime(CLOCK_REALTIME,NULL);
        }''',

In my case, I was calling clock_gettime inside the handler. And i was using 
clock_gettime(CLOCK_REALTIME,&real_time_clock); instead of NULL. I assume this is why it wasn't requiring it. Thus not being able to compile.

So I just changed Mandatory to True and although it kept saying "not necesary" in the configuration step, it did compile and gave correct behaviour.

I hope this helps to anyone.

Regards,
Oscar

OXINARF, 2017-09-06:
This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

closed 2017-09-06 by OXINARF