ArduPilot/ardupilot #6903: Feature Request: logical switches.

geofrancis, 2017-09-05:
if you have used opentx then you have no doubt seen the power of logical switches to do things that you cant do on other systems, things like 

landing gear going up and down depending on altitude.
led flash and pulse sequences for gps status.
stall recovery switch that switched to acro and stopped rotation before switching to fbwa.
extra failsafe that switches to land mode if critical voltage is hit during RTL,

if there was logical switches in ardupilot that can be activated by any value from the flight controller a lot of new features could be added just by setting some values, it could also allow a lot of customisation of existing features if they were broken out into logical switches, things like failsafe actions so any value could trigger RTL from current too high to the board being too cold.

sashan13, 2017-09-06:
+++
Totally agree. RC input seems so inflexible at moments. Also, features like auto start recording when armed and stop when disarmed would be easy to implement.

geofrancis, 2017-09-08:
the sky is the limit with logical switches, once you have them you have everything you need to make a sequencer, you could probably make it walk :)

Naterater, 2018-12-01:
I think the new RC_OPTIONS features cover this topic. It's not exactly the same, but it's close enough to consider this closed.

[image]

closed 2018-12-01 by Naterater