ArduPilot/ardupilot #6907: Camera Trigger - add 3 positions support.

sashan13, 2017-09-06:
#### Issue details
So, I've got myself a nice trigger module that supports dozens of cameras. I use it for Sony a7. It has 2 inputs for triggering, one triggers photo when high, starts/stops video when low, does nothing at 1.5ms.  

Second one is 3 positions, akin to camera's hardware trigger button. Low state is normal, put it to mid, camera focuses, high - take photo.   The problem is, if I give this channel to Ardupilot it doesn't work since camera won't take photo when Ardupilot jumps from lowest to highest position while skipping the middle one. If I give first channel to Ardupilot I lose ability to record videos.

I propose adding option of 2 stage trigger. Add two more parameters, like CAM_MID_DURATION and CAM_SERVO_MID.
#### Version
3.5.2
#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Any

#### Hardware type
Pixhack

#### Logs
No logs

geofrancis, 2017-09-06:
related to https://github.com/ArduPilot/ardupilot/issues/6248 ?

sashan13, 2017-09-06:
Yes.

OXINARF, 2017-09-06:
Duplicate of #6248

closed 2017-09-06 by OXINARF