ArduPilot/ardupilot #6913: [Bug] NSH sending nsh> \0x1b[K every line

whyscience, 2017-09-07:
The latest master, here is the bug

- For cutecom
[image]

- For SecureCRT
[image]

> nsh> \0x1b[K
nsh> \0x1b[K
nsh> \0x1b[K
nsh> \0x1b[K
nsh> \0x1b[K
nsh> \0x1b[K
nsh> \0x1b[Kls
/:
 bin/
 dev/
 etc/
 fs/
 obj/
nsh> \0x1b[K

whyscience, 2017-09-07:
There is no this bug on ArduCopter v3.5

peterbarker, 2017-09-07:
Check your terminal settings emulation settings.  This is a valid escape sequence, so it should be being displayed.

whyscience, 2017-09-07:
However there is no such character before.
And secureCRT is Not normal

kingdwd, 2017-09-07:
you should  delet     ‘“ hal.uartF->begin(57600); ” in function "static int main_loop(int argc, char **argv) " in HAL_PX4_Class.cpp

VikingDai, 2017-09-07:
You should delete “ hal.uartF->begin(57600); ” in function "static int main_loop(int argc, char **argv) " in HAL_PX4_Class.cpp, the driver "ttys5"  reinited

OXINARF, 2017-09-07:
Like @peterbarker said, this is an ASCII escape sequence, you need to check your terminal settings to allow it, this isn't a bug.

closed 2017-09-07 by OXINARF