ArduPilot/ardupilot #6918: Add new waypoints in auto mode

medhijk, 2017-09-07:
Using the instructions for setting up ardupilot in [http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html#setting-up-sitl-on-linux](url), I loaded the waypoint file in the terminal as `wp load ../Tools/autotest/ArduPlane-Missions/CMAC-toff-loop.txt`. I did `mode auto' and then `arm throttle` from the same terminal and the plane is following the waypoints correctly. However, I noticed that after completing all the waypoints, the plane changes to RTL mode following which it goes back to where it was armed before and starts circling around it. For my new project, I want the plane to stay on the last waypoint after reaching that last waypoint (it may hover or land) and wait for further waypoints. If further waypoints are provided, the plane should follow to those immediately, otherwise just stay on the last waypoint. The problem I'm having now is the plane changes to RTL after mission completion and when I add new waypoints  to the waypoint textfile and load it, the plane starts the whole mission from the beginning.

Any idea how to resolve this. Is it possible to do this by editing the mav_cmd commands and parameters in the waypoint textfile?

OXINARF, 2017-09-07:
This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

You'll need to change code to do that. It's not a very hard change but not an easy one either. The mission library is coded to complete once there are no more commands and that will call https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/commands_logic.cpp#L989, which are the lines that set RTL when the mission ends.

closed 2017-09-07 by OXINARF

rrr6399, 2017-09-08:
@medhijk , You could always make the last step of the mission unlimited loiter and then switch to guided mode and send guided waypoints. I imagine you could even make the last step of the mission switch to guided mode and then send guided waypoints.